ethz-asl/kalibr #178: How long should calibration samples should be?

JJQ7, 2018-04-10:
Hi, when looking Camera-IMU calibration tutorial, I notice:

>The calibration target is fixed in this calibration and the camera-imu system is moved in front of the target to excite all IMU axes. It is important to ensure good and even illumination of the calibration target and to keep the camera shutter times low to avoid excessive motion blur.

However, how long should calibration samples should be? Does the sample size has an effect on the calibration result? Hope to get some advice.

Thanks!

ZacharyTaylor, 2018-04-10:
When I do the calibration my bags are usually between 30 seconds and 2 minutes.

As long as all axes are excited you should get accurate results, but as a rule of thumb the longer the bag the higher the accuracy as the sensor noise has less of an impact on results. However, you hit diminishing returns on this pretty quickly.

NikolausDemmel, 2018-04-10:
We usually do 40-60 second sequences for cam-imu calibration with Kalibr.

For us the typical motion is (all together is 40-60 secs): 
1. accelerations: UP-DOWN, LEFT-RIGHT, FORWARD-BACK, 3x each
2. rotations: roll LEFT-RIGHT, pitch UP-DOWN, pan LEFT-RIGHT, 3x each
3. some smooth motion with all DOG, e.g. infinity sign while rotating a bit

What is important:
1. Pattern is always visible
2. No shocks to avoid accelerometer spikes; just smooth motion
3. Low exposure times (good lighting) to minimize motion blur

If you take much longer sequences, calibration will also take much longer to compute with questionable benefit.

JJQ7, 2018-04-11:
Great thanks. That's really helpful.

closed 2022-06-02 by goldbattle