ethz-asl/kalibr #180: Invalid Tranform IMU to CAM

jsonvillanueva, 2018-04-19:
To produce a neuroc.yaml for use with Maplab, the camera and IMU must be calibrated; thus I used Kalibr for this. As I'm currently only using a single front facing camera on a [Jackal](https://www.clearpathrobotics.com/jackal-small-unmanned-ground-vehicle/), the multiple camera calibration was used to calibrate a single camera's intrinsic properties:
camchain-[homeuser]bags[date].yaml

Using this single camera in the camchain, I did an IMU-Camera calibration (noise densities/random walk in pasted below). I've pasted the results of imucam calibration (report-imucam-[home+user]bags[date].txt). Unfortunately, the bags I used ([bag1](https://www.dropbox.com/s/w1ot2pb7iz6gb1z/2018-03-15-09-22-41.bag?dl=0), [bag2](https://www.dropbox.com/s/610vhadlukrdvxr/2018-03-15-10-19-47.bag?dl=0)) in the Kalibr framework produce a T_B_C matrix that is invalid in the Maplab framework. I'm assuming the base is the imu in T_B_C.

```
Calibration results
===================
Normalized Residuals
----------------------------
Reprojection error (cam0):     mean 0.571760641483, median 0.378809808438, std: 0.843036952643
Gyroscope error (imu0):        mean 0.18778797698, median 0.0357102803091, std: 0.431251481723
Accelerometer error (imu0):    mean 0.0199465992241, median 0.00215364678309, std: 0.0526314440173

Residuals
----------------------------
Reprojection error (cam0) [px]:     mean 0.571760641483, median 0.378809808438, std: 0.843036952643
Gyroscope error (imu0) [rad/s]:     mean 0.0046024051881, median 0.000875206080848, std: 0.0105693351022
Accelerometer error (imu0) [m/s^2]: mean 0.00500953587769, median 0.000540882719232, std: 0.0132182485915

Transformation (cam0):
-----------------------
T_ci:  (imu0 to cam0): 
[[-0.00857888 -0.99989257 -0.01188529 -0.31239235]
 [ 0.25528736  0.00930187 -0.96682048  0.05485438]
 [ 0.96682717 -0.0113284   0.25518013 -1.55120937]
 [ 0.          0.          0.          1.        ]]

T_ic:  (cam0 to imu0): 
[[-0.00857888  0.25528736  0.96682717  1.48306776]
 [-0.99989257  0.00930187 -0.0113284  -0.33044175]
 [-0.01188529 -0.96682048  0.25518013  0.44515927]
 [ 0.          0.          0.          1.        ]]

timeshift cam0 to imu0: [s] (t_imu = t_cam + shift)
0.0

Gravity vector in target coords: [m/s^2]
[ 0.49928714 -9.62739887 -1.79786714]

Calibration configuration
=========================

cam0
-----
  Camera model: pinhole
  Focal length: [602.0860627331539, 600.3129218871015]
  Principal point: [326.3289311430195, 236.5897837327919]
  Distortion model: radtan
  Distortion coefficients: [0.111069717816489, -0.26906846531794615, -0.004683258781562304, -0.0009927168836868477]
  Type: aprilgrid
  Tags: 
    Rows: 6
    Cols: 6
    Size: 0.088 [m]
    Spacing 0.0264 [m]

IMU configuration
=================

IMU0:
----------------------------
  Model: calibrated
  Update rate: 75.0
  Accelerometer:
    Noise density: 0.029 
    Noise density (discrete): 0.251147367097 
    Random walk: 0.006
  Gyroscope:
    Noise density: 0.00283
    Noise density (discrete): 0.0245085189271 
    Random walk: 0.000225
  T_i_b
    [[ 1.  0.  0.  0.]
     [ 0.  1.  0.  0.]
     [ 0.  0.  1.  0.]
     [ 0.  0.  0.  1.]]
  time offset with respect to IMU0: 0.0 [s]
```

ZacharyTaylor, 2018-04-20:
When you say it is invalid for maplab is it throwing an error or simply giving wrong results. If it is an error could you copy both that error and the calibration yaml files you are passing to maplab here.

jsonvillanueva, 2018-04-24:
It throws an error. Here's the resulting error when attempting to use maplab (where mapit is the script that runs rovio from a bag file/topic): 
```jason@acl-super src/maplab ➤ ./mapit                                                      git:master* │                                                                                                     
F0419 19:44:39.417974  2003 rotation-quaternion-inl.h:115] Check failed: isValidRotationMatrix(matrix)│                                                                                                     
 000000000 00.267488 00.963561                                                                        │                                                                                                     
0000000-1 000000000 000000000                                                                         │                                                                                                     
000000000 -0.963561 00.267488                                                                         │                                                                                                     
*** Check failure stack trace: ***                                                                    │                                                                                                     
    @     0x7f65851bc26d  google::LogMessage::Fail()                                                  │                                                                                                     
    @     0x7f65851be083  google::LogMessage::SendToLog()                                             │                                                                                                     
    @     0x7f65851bbdfb  google::LogMessage::Flush()                                                 │                                                                                                     
    @     0x7f65851bea6e  google::LogMessageFatal::~LogMessageFatal()                                 │                                                                                                     
    @     0x7f6585add3e9  YAML::convert<>::decode()                                                   │                                                                                                     
    @     0x7f6585ace67a  aslam::NCamera::deserializeFromYaml()                                       │                                                                                                     
    @     0x7f6585ace8f3  aslam::NCamera::loadFromYaml()                                              │                                                                                                     
    @           0x40b1d0  main                                                                        │                                                                                                     
    @     0x7f6583d19830  __libc_start_main                                                           │                                                                                                     
    @           0x40bf09  _start                                                                      │                                                                                                     
./mapit: line 55:  2003 Aborted                 (core dumped) rosrun rovioli rovioli --alsologtostderr│                                                                                                     
=1 --v=2 --ncamera_calibration=$NCAMERA_CALIBRATION --imu_parameters_maplab=$IMU_PARAMETERS_MAPLAB --i│                                                                                                     
mu_parameters_rovio=$IMU_PARAMETERS_ROVIO --datasource_type="rosbag" --save_map_folder="$LOCALIZATION_│                                                                                                     
MAP_OUTPUT" --optimize_map_to_localization_map=false --map_builder_save_image_as_resources=false --dat│                                                                                                     
asource_rosbag=$ROSBAG $REST
```
*Note noise densities/random walks I've increased by an order of magnitude here. I calculated these parameters directly from sensor data.

**$IMU_PARAMTERS_MAPLAB**:
```
hardware_id: /JA01/imu/data_raw
sensor_type: IMU
sigmas:
  gyro_noise_density: 0.00283
  gyro_bias_random_walk_noise_density: 2.25e-04
  acc_noise_density: 0.029
  acc_bias_random_walk_noise_density: 0.006
saturation_accel_max_mps2: 150.0
saturation_gyro_max_radps: 7.5
gravity_magnitude_mps2: 9.81
update_rate:                 75.0      #Hz (for discretization of the values above)
```

**$IMU_PARAMETERS_ROVIO**:
```
acc_noise_density: 2.9e-01
acc_bias_random_walk_noise_density: 6.0e-02
gyro_noise_density: 2.83e-02
gyro_bias_random_walk_noise_density: 2.25e-03
```
*Note: I've changed the T_B_C matrix to something reasonable (as the previous post shows a T_B_C with a linear translation in the x-direction of 1.4 and none of the Jackal's dimension are that large). The same error results with the IMU/Cam0 transform calculated from Maplab and a rough estimate with a ruler.
**$NCAMERA_CALIBRATION**:
```label: ACL Single Camera
id: aad0e0b609f332d5e42f0fbacc0e75cb
cameras:
  - camera:
      label: /realsense_front_middle/color
[... cut ...]
      distortion:
        type: radial-tangential
        parameters:
          rows: 4
          cols: 1
          data:
            - 0.111069717816489
            - -0.26906846531794615
            - -0.004683258781562304
            - -0.0009927168836868477
    *T_B_C:  #(cam0 to imu0): 
      cols: 4
      rows: 4
      data: [0.0,  0.2674882,  0.9635612,  .20,
	      -1.0,  0.0, 0.0 , 0.0,
	       0.0, -0.9635612,  0.2674882,  0.02,
	       0.        ,  0.        ,  0.        ,  -.001]
```

ZacharyTaylor, 2018-04-24:
The issue is that you seem to have lost some sig-figs along the way in getting your T_B_C in the NCAMERA_CALIBRATION file. Maplab uses minkindr behind the scenes and a slightly frustrating feature of minkindr is that any rotation matrix provided to it must confirm to all the properties of a rotation matrix to a tolerance that sometimes feels like its a hairs breadth from machine precision.

In your example you can fix it by changing data to:
[2.22044604925031e-16 ,   0.267488138357663,    0.963561153138685 ,                 0.2,
                  -1.0,                    0.0, 1.66533453693773e-16,                    0.0,
                   0.0,   -0.963561153138685,    0.267488138357663,                 0.02,
                   0.0,                    0.0,                    0.0,                    1.0]

jsonvillanueva, 2018-04-25:
So the precision issue comes from taking the values for T_imu_camera in `results-imucam-xxxxx.txt` as opposed to `camchain-imucam-xxxx.yaml`. This may be something worth adding into the documentation.

Something else I recently noted is that using the `--time-calibration` flag in `kalibr_calibrate_imu_camera` produces a T_B_C matrix that is much closer the ruler calibration; however, all three calibrations: normal kalibr_imu_cam calibration; flag kalibr_imu_cam calibration; ruler calibration, have faults in landmark recognition in rovio and get spastic as the Jackal robot turns around our lab's office.

ZacharyTaylor, 2018-04-26:
A significant offset in time can result in poor calibration parameters. If you don't compensate for this discrepancy between the given timestamps and the real time then this can also cause rovio to fail. How large is the reported time offset as anything more than 20 ms or so will probably be an issue.

jsonvillanueva, 2018-04-27:
The time shift between imu0/cam0 is 0.00143248193163 which is safely under 20ms

ZacharyTaylor, 2018-05-06:
Going on my experience digging around in the D435 driver I think it could be at least part of your issue. We currently have a pr open on our fork where I hacked in some timesync using our time translator library. https://github.com/ethz-asl/realsense/pull/28

Razbotics, 2019-08-05:
Hey Zachary, Could you tell me if intel D435 can be hardware synchronised with external IMU using those HW pins using external triggering?

Pan-long-ming, 2021-11-15:
I also met this problem
 "F1115 15:24:24.066972 21321 rotation-quaternion-inl.h:115] Check failed:
isValidRotationMatrix(matrix) 00.0207655 0-0.999881 00.0041403
000.999557 00.0149672 00.0257155
-0.0257744 0.00375619 000.999661
"
How did you solve it,can you give me some suggestion?

goldbattle, 2022-06-02:
Closing due to age.