ethz-asl/kalibr #181: Camera IMU calibration

firelw, 2018-04-26:
question 1:
The process of camera calibration i use，is it right?
1. make calibration board,
2. collect image and make bag, then camera calibration,
3. get the white noise and bias using the datasheet of imu，or get the noise and bias using Allan tool，
4. make target.yaml,camera.yaml,imu.yaml.
5. collect image and imu data, make bag,then camera-imu calibration.
----------------------------------------------------------------------------------------
question 2:
is it need to calibrate the imu before camera-imu calibration?

yangtzey, 2018-09-06:
it is need to  calibrate the imu before camera-imu calibration. i use https://github.com/gaowenliang/imu_utils to calibration imu.

goldbattle, 2022-06-02:
Closing due to age.

closed 2022-06-02 by goldbattle

yangtzey, 2022-06-02:
这是来自QQ邮箱的假期自动回复邮件。
自动回复