ethz-asl/kalibr #183: Monocular Camera IMU calibration

SubMishMar, 2018-04-28:
Can we perform a monocular camera imu calibration using this package? The tutorial given in the wiki is just for stereo imu calibration.

NikolausDemmel, 2018-04-28:
I guess it should work just the same. Have you tried?

SubMishMar, 2018-04-28:
Okay! No I havent tried because the tutorial says that I need to send both left and right camera images. I will try and find out then.

SubMishMar, 2018-04-29:
Yeah it seems to work with a monocular cam+imu setup.. Thanks!

SubMishMar, 2018-04-30:
In the yaml file for april tags, there are parameters like : 

tagSize: 0.088           #size of apriltag, edge to edge [m]
tagSpacing: 0.3          #ratio of space between tags to tagSize

Does tagSize mean the length of a side of the apriltag?

schneith, 2018-05-02:
Yes, tagSize should be the length of the tag, edge-to-edge.

SubMishMar, 2018-05-02:
Cool! Thanks!

closed 2022-04-27 by goldbattle