ethz-asl/kalibr #187: Could not read configuration from checkerboard.yaml

kennege, 2018-05-21:
I have tried using the Kalibr-CDE in both Ubuntu 14.04 and 16.04. 

When I run the multi camera calibration with the following line:

$ ./kalibr_calibrate_cameras --bag awesome.bag --topics /cam0/image_raw /cam1/image_raw /cam2/image_raw /cam3/image_raw /cam4/image_raw /cam5/image_raw --models pinhole-equi pinhole-equi pinhole-equi pinhole-equi pinhole-equi pinhole-equi --target checkerboard.yaml

I get the following error message:

RuntimeError: [CalibrationTargetConfig Reader]: Could not read configuration from checkerboard.yaml
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/apport_python_hook.py", line 64, in apport_excepthook
    from apport.fileutils import likely_packaged, get_recent_crashes
ImportError: No module named apport.fileutils

The .bag and .yaml files are then also deleted.

Has anyone had a similar problem, or might know what is causing this issue?

ravikt, 2018-08-09:
I am also having the similar problem

BmlmnnsDev, 2019-02-11:
Same Problem here

Heymor, 2020-08-06:
I had the same problem, including the `.bag` and `.yaml` files being deleted. I solved it using the absolute path to the datasets, and making sure that the `.bag` and `.yaml` files are NOT in the kalibr-cde directory.

zhengjialiang2, 2021-01-11:
> This is an old issue, but I'll put this here for anyone coming across this now.
> 
> I had the same problem when using the CDE install, including the `.bag` and `.yaml` files being deleted. I solved it using the absolute path to the datasets, and making sure that the `.bag` and `.yaml` files are NOT in the kalibr-cde directory.

Thank.failed to compile source code and meet the same problem running cde with ubuntu 18.04, and this method really work for me.

closed 2022-06-02 by goldbattle