ethz-asl/kalibr #189: how to use scale-misalignment result?

Neil-Oyoung, 2018-05-24:
Hello, after I used kalibr_calibrate_imu_camera with scale-misalignment option, I got following result. 
Do anyone know how to use *M, A or C_gyro_i* to correct IMU raw measurements ?

```
IMU configuration
=================

IMU0:
----------------------------
  Model: scale-misalignment
  Update rate: 500
  Accelerometer:
    Noise density: 0.039813853765 
    Noise density (discrete): 0.890264834649 
    Random walk: 0.00347566152835
  Gyroscope:
    Noise density: 0.00861190257109
    Noise density (discrete): 0.192567995646 
    Random walk: 0.000577879736374
  T_i_b
    [[ 1.  0.  0.  0.]
     [ 0.  1.  0.  0.]
     [ 0.  0.  1.  0.]
     [ 0.  0.  0.  1.]]
  time offset with respect to IMU0: 0.0 [s]
  Gyroscope: 
    M:
      [[ 0.99732203  0.          0.        ]
       [ 0.00373207  0.99372249  0.        ]
       [-0.00218998 -0.00186741  0.99919668]]
    A [(rad/s)/(m/s^2)]:
      [[-0.00075474  0.00162226  0.00019704]
       [-0.00346093 -0.00023962 -0.00694434]
       [-0.00216607  0.00254173  0.00037087]]
    C_gyro_i:
      [[ 0.99999949  0.00071135  0.00071417]
       [-0.00071004  0.99999806 -0.00183835]
       [-0.00071547  0.00183784  0.99999806]]
  Accelerometer: 
    M:
      [[ 0.98064672  0.          0.        ]
       [-0.01260154  1.01224075  0.        ]
       [-0.01087295  0.00237432  1.00600802]]
```

QingfengLi-hit, 2018-09-03:
reference paper Extending kalibr: Calibrating the Extrinsics of Multiple IMUs and of Individual Axes

AmosLewis, 2019-04-11:
> reference paper Extending kalibr: Calibrating the Extrinsics of Multiple IMUs and of Individual Axes

The paper doesnot explain it well. If the M here is M or S*M, weather it contains scale in it or not?

arenas7307979, 2020-12-15:
same question, How to correct misalignment and scale through these parameters

mhyoosefian, 2021-11-25:
Although this issue is for a long time ago, I put the answer here, for future reference:

Say you have a set of raw measurements:
<img src="https://render.githubusercontent.com/render/math?math=\boldsymbol{\omega}"> , <img src="https://render.githubusercontent.com/render/math?math=\mathbf{a}">.
From the Kalibr you have gyroscope misalignment and g-sensitivity matrices: 
<img src="https://render.githubusercontent.com/render/math?math=\textbf{M}_{g}">, <img src="https://render.githubusercontent.com/render/math?math=\textbf{A}_{g}">
, accelerometer misalignment matrix: 
<img src="https://render.githubusercontent.com/render/math?math=\textbf{M}_{a}">
, and the rotation matrix from gyroscope to accelerometer: 
<img src="https://render.githubusercontent.com/render/math?math=\textbf{C}_{g a}">.
Firstly, you have to transfer gyroscope reading to the accelerometer coordinate frame (i.e., the body frame):
<img src="https://render.githubusercontent.com/render/math?math=\boldsymbol{\omega}^{a} = \textbf{C}_{g a}^{-1}\boldsymbol{\omega}">.
Next, you apply the matrices using the following formulas:
<img src="https://render.githubusercontent.com/render/math?math=\boldsymbol{\omega}^{new} = \textbf{M}_{g}^{-1}(\boldsymbol{\omega}^{a} - \textbf{A}_{g}\textbf{a})">
<img src="https://render.githubusercontent.com/render/math?math=\textbf{a}^{new} = \textbf{M}_{a}^{-1}\textbf{a}">
These are the new (i.e., the corrected) measurements.
Although these formulas are obtainable from the reference paper, to be over-sure (!), after calibrating my setup, I have corrected the calibration measurements and reran the Kalibr using the corrected measurements. The new matrices were so close to their ideal values (Identity for misalignments and Zero for g-sensitivity). However, I recommend you doing the same and report any issues!

Regards,
Mohammad

goldbattle, 2022-04-22:
This paper has details of the different models:
https://timohinzmann.com/publications/icra_2016_rehder.pdf

For example you can see the following:
[image]

Additionally one can inspect the code to compare:
https://github.com/ethz-asl/kalibr/blob/master/aslam_offline_calibration/kalibr/src/GyroscopeError.cpp#L70-L73

This paper also has some background details if you want to further research:
https://roboticsconference.org/2020/program/papers/26.html

closed 2022-06-14 by goldbattle

sgraybi, 2023-04-04:
After working through the [paper](https://timohinzmann.com/publications/icra_2016_rehder.pdf), the [code](https://github.com/ethz-asl/kalibr/blob/master/aslam_offline_calibration/kalibr/src/GyroscopeError.cpp#L70-L73), and the output yaml files I'm still not certain that I have the correct formulation to correct raw accelerometer and raw gyro measurements.  It's not trivial to reconcile the three as they use different notations. A small error in applying the calibration may not be easily detectable in the calibrated result.  The text below aims to reconcile the notation between the [paper](https://timohinzmann.com/publications/icra_2016_rehder.pdf) and the Kalibr output yaml files.  The notation below should match the [paper](https://timohinzmann.com/publications/icra_2016_rehder.pdf).  @goldbattle please verify for correctness.

### Coordinate frames
$B$: body frame
$W$: world/calibration board frame
$A$: IMU frame
$C$: Camera frame

### Coordinate frame transformation

$C\_{XY}$ denotes the direction cosine matrix, rotation matrix, that rotates a vector from frame $Y$ to frame $X$, i.e., $x\_X = C\_{XY} x\_Y$

### Parameter definition from Kalibr IMU calibration file

Let's define the parameters in the XXX-imu.yaml files as follows:

$C^\alpha\_{AB}\equiv$ `T_i_b` (4x4 double matrix) [Pose matrix that transforms an accelerometer measurement in the body frame, $B$, to the IMU frame, $A$.]

$S\_{\alpha}M\_{\alpha} \equiv$ `accelerometers.M` [1] (3x3 lower triangular double matrix) [Applies the accelerometer scale factor and misalignment correction.]

$A\_{\omega} \equiv$ `gyroscopes.A` [rad.s^-1 / (m.s^{-2})] (3x3 double matrix) [Effect of linear accelerations on gyroscopes.]

$C^{\omega}\_{AB} \equiv$ `gyroscopes.C_gyro_i` [1] (3x3 double matrix) [Rotation matrix that rotates a gyroscope measurement in the body frame, $B$, to the IMU frame, $A$.]

$S\_{\omega}M\_{\omega} \equiv$ `gyroscopes.M` [1] (3x3 lower triangular double matrix) [Applies the gyroscope scale factor and misalignment correction.]

### Parameter definition from Kalibr camera calibration file

Let's define the parameter in the XXX-camchain-imucam.yaml files as follows:

$C\_{CA} \equiv$ `cam0.T_cam_imu` (4x4 double matrix) [Pose matrix that transforms a vector in the IMU frame, $A$, to the camera frame, $C$.]

### How to apply intrinsic calibration

Let's denote the uncalibrated, raw, accelerometer and gyroscope measurements as $\alpha$ and $\omega$, respectively.  The calibrated accelerometer and gyroscope measurements in the body frame, $B$, are then given by:

$$ \alpha\_B= (C^{\alpha}\_{AB})^{T} (S\_\alpha M\_\alpha)^{-1} (\alpha - b\_\alpha)$$

$$ \omega\_B= (C^{\omega}\_{AB})^{T} (S\_\omega M\_\omega)^{-1} (\omega -A\_\omega \alpha\_A - b\_\omega)$$

Note well, $b\_\alpha$ and $b\_\omega$ are the accelerometer and gyroscope biases.  These parameters are **NOT** stored in the IMU or camera yaml files but are included here for completeness.

Using the same notation and parameters as the yaml files

$\alpha\_B = ($`T_i_b`$)^{T}($`accelerometers.M`$)^{-1} \alpha$

$\omega\_B = ($`gyroscopes.C_gyro_i`$)^{T}($`gyroscopes.M`$)^{-1}(\omega - $`gyroscopes.A`$\alpha\_A)$

The accelerometer and gyro biases are not applied here as they are not stored in the yaml files.

### How to apply extrinsic calibration

For some applications the IMU measurement needs to be rotated into the camera frame, $C$.  This can be done using:

$\alpha\_C = C\_{CA}\alpha\_A = $`cam0.T_cam_imu`$\alpha\_A$

$\omega\_C = C\_{CA}\omega\_A = $`cam0.T_cam_imu`$\omega\_A$

Note well that this is only true in the static case for the accelerometer.  See equations (1), (2), and (5) from [paper](https://timohinzmann.com/publications/icra_2016_rehder.pdf) for the dynamic case.

[githubreply.md](https://github.com/ethz-asl/kalibr/files/11151462/githubreply.md)

cai199626, 2023-04-07:
> After working through the [paper](https://timohinzmann.com/publications/icra_2016_rehder.pdf), the [code](https://github.com/ethz-asl/kalibr/blob/master/aslam_offline_calibration/kalibr/src/GyroscopeError.cpp#L70-L73), and the output yaml files I'm still not certain that I have the correct formulation to correct raw accelerometer and raw gyro measurements. It's not trivial to reconcile the three as they use different notations. A small error in applying the calibration may not be easily detectable in the calibrated result. The text below aims to reconcile the notation between the [paper](https://timohinzmann.com/publications/icra_2016_rehder.pdf) and the Kalibr output yaml files. The notation below should match the [paper](https://timohinzmann.com/publications/icra_2016_rehder.pdf). @goldbattle please verify for correctness.
> 
> ### Coordinate frames
> B: body frame W: world/calibration board frame A: IMU frame C: Camera frame
> 
> ### Coordinate frame transformation
> CXY denotes the direction cosine matrix, rotation matrix, that rotates a vector from frame Y to frame X, i.e., xX=CXYxY
> 
> ### Parameter definition from Kalibr IMU calibration file
> Let's define the parameters in the XXX-imu.yaml files as follows:
> 
> CABα≡ `T_i_b` (4x4 double matrix) [Pose matrix that transforms an accelerometer measurement in the body frame, B, to the IMU frame, A.]
> 
> SαMα≡ `accelerometers.M` [1] (3x3 lower triangular double matrix) [Applies the accelerometer scale factor and misalignment correction.]
> 
> Aω≡ `gyroscopes.A` [rad.s^-1 / (m.s^{-2})] (3x3 double matrix) [Effect of linear accelerations on gyroscopes.]
> 
> CABω≡ `gyroscopes.C_gyro_i` [1] (3x3 double matrix) [Rotation matrix that rotates a gyroscope measurement in the body frame, B, to the IMU frame, A.]
> 
> SωMω≡ `gyroscopes.M` [1] (3x3 lower triangular double matrix) [Applies the gyroscope scale factor and misalignment correction.]
> 
> ### Parameter definition from Kalibr camera calibration file
> Let's define the parameter in the XXX-camchain-imucam.yaml files as follows:
> 
> CCA≡ `cam0.T_cam_imu` (4x4 double matrix) [Pose matrix that transforms a vector in the IMU frame, A, to the camera frame, C.]
> 
> ### How to apply intrinsic calibration
> Let's denote the uncalibrated, raw, accelerometer and gyroscope measurements as α and ω, respectively. The calibrated accelerometer and gyroscope measurements in the body frame, B, are then given by:
> 
> αA=(SαMα)−1(α−bα)
> 
> ωA=(SωMω)−1(ω−AωαA−bω)
> 
> αB=(CABα)TαA
> 
> ωB=(CABω)TωA
> 
> Note well, bα and bω are the accelerometer and gyroscope biases. These parameters are **NOT** stored in the IMU or camera yaml files but are included here for completeness.
> 
> Using the same notation and parameters as the yaml files
> 
> αA=(`accelerometers.M`)−1α
> 
> ωA=(`gyroscopes.M`)−1(ω−`gyroscopes.A`αA)
> 
> αB=(`T_i_b`)TαA
> 
> ωB=(`gyroscopes.C_gyro_i`)TωA
> 
> The accelerometer and gyro biases are not applied here as they are not stored in the yaml files.
> 
> ### How to apply extrinsic calibration
> For some applications the IMU measurement needs to be rotated into the camera frame, C. This can be done using:
> 
> αC=CCAαA=`cam0.T_cam_imu`αA
> 
> ωC=CCAωA=`cam0.T_cam_imu`ωA
> 
> Note well that this is only true in the static case for the accelerometer. See equations (1), (2), and (5) from [paper](https://timohinzmann.com/publications/icra_2016_rehder.pdf) for the dynamic case.
> 
> [KalibrHowToApplyCalibration.md](https://github.com/ethz-asl/kalibr/files/11151801/KalibrHowToApplyCalibration.md)

@sgraybi thanks for your answer, but your formulas are not equal to @mhyoosefian above, I do not know who is right...

sgraybi, 2023-04-07:
@cai199626 yes.  We need @goldbattle, or another expert to review.  Intuitively, you would apply the intrinsics before the extrinsics, like in my correction equations.  Intuitively, you would also correct the accelerometer measurement before it's used to correct the effect of linear accelerations on gyroscopes.  I believe the mathematical formulation supports these intuitions also.

I'm don't quite understand how other people know how to apply the model.  Kalibr seems to be quite widely used.