ethz-asl/kalibr #190: bagcreater: bag has duration 0s and thus can only calibrate one image

HenryZh47, 2018-05-28:
I have a stream of images recorded from my camera and I created a ros bag with the bagcreater. However the bag info shows that the duration is 0 second and thus can't be successfully calibrated.
I have only one camera needs to be calibrated.
Are there certain ways that I need to name the file before creating the bag? or anything else I need to do?

Thank you!

NikolausDemmel, 2018-05-28:
I don't know about bagcreator, but just a note on the 0 duration:

Does the bag actually contain all the images, or is it empty? If it is the former, the duration=0 might be related to this rosbag issue: https://github.com/ros/ros_comm/issues/952

HenryZh47, 2018-05-28:
I solved by renaming all the image files. It seems to me that the script creates bag timestamp info based on the file names of the images. By changing the file names to series of numbers with an increment on the 8th bit makes the timestamp increases 0.1s per image.

This is quite strange for me. Not sure if it is the right way to explain but it works for me.

NikolausDemmel, 2018-05-28:
I guess that makes sense and you can probably close this issue.

closed 2018-05-28 by HenryZh47