ethz-asl/kalibr #192: parameters ambiguity of aprilgrid.yaml

ZhengPan2014, 2018-06-07:
@kartikmohta @skohlbr @NikolausDemmel @alexmillane 
any detail reference to the parameters in aprilgrid.yaml file, especially the last two. thank you for detail reference.
```
tagSize: 0.088           #size of apriltag, edge to edge [m]
tagSpacing: 0.3          #ratio of space between tags to tagSize
```

schneith, 2018-06-14:
tagSize is the size of Apriltag; edge to edge [m].
tagSpacing is the (unitless) ratio of space between tags [m] and the tagSize [m] (as defined above)

ZhengPan2014, 2018-06-15:
@schneith 
just like this?
[image]

schneith, 2018-06-15:
yep, that looks correct

ZhengPan2014, 2018-06-15:
of course, glad about that.

closed 2022-04-27 by goldbattle