ethz-asl/kalibr #194: apparent error of T from imu to camera

ZhengPan2014, 2018-06-13:
@hitzg 
any ideal to improve the result of calibration, especially translation between camera and imu. I got the below result with my mynteye VI device?

```
After Optimization (Results)
==================
Normalized Residuals
----------------------------
Reprojection error (cam0):     mean 0.175634891833, median 0.124722507318, std: 0.223894089103
Reprojection error (cam1):     mean 0.195597579339, median 0.148223783893, std: 0.414186213603
Gyroscope error (imu0):        mean 0.287852891578, median 0.226832007742, std: 0.247774302859
Accelerometer error (imu0):    mean 0.177563015623, median 0.137547220363, std: 0.183814648925

Residuals
----------------------------
Reprojection error (cam0) [px]:     mean 0.175634891833, median 0.124722507318, std: 0.223894089103
Reprojection error (cam1) [px]:     mean 0.195597579339, median 0.148223783893, std: 0.414186213603
Gyroscope error (imu0) [rad/s]:     mean 0.0295382864363, median 0.023276572908, std: 0.0254255855804
Accelerometer error (imu0) [m/s^2]: mean 0.0561503557577, median 0.0434962502173, std: 0.0581272957908

Transformation T_cam0_imu0 (imu0 to cam0, T_ci): 
[[-0.00384462  0.99957543  0.02888214  0.18148842]
 [-0.99929742 -0.00276357 -0.03737692 -0.03371517]
 [-0.03728124 -0.02900555  0.99888377 -0.32026818]
 [ 0.          0.          0.          1.        ]]

Transformation T_cam1_imu0 (imu0 to cam1, T_ci): 
[[-0.00200624  0.99957442  0.0291026   0.060636  ]
 [-0.99924054 -0.00087135 -0.03895627 -0.03282727]
 [-0.03891434 -0.02915866  0.99881702 -0.32110943]
 [ 0.          0.          0.          1.        ]]

```

schneith, 2018-06-14:
We had some issues with the data from this sensor when we last tried to calibrate it. Maybe have a look at this page and update the firmware if necessary: https://github.com/slightech/MYNT-EYE-SDK/issues/22. 

Also make sure that the dataset provides enough excitation along all the IMU axis and that there is not too much motion blur.

Chuck-Yu, 2018-10-23:
Maybe you can try to add the `--time-calibration` flag when run the `kalibr_calibration_imu_camera`. #180

goldbattle, 2022-06-02:
Closing due to age.

closed 2022-06-02 by goldbattle