ethz-asl/kalibr #199: variable camera-imu time lag

redbehemoth, 2018-06-27:
Hi,
We've been running some experiments on a custom system and found that after Kalibr camera-imu calibration completes, we are seeing in the same sequence times where 1) IMU signal leads camera signal, 2) IMU signal lags behind camera signal, and 3) IMU/camera signals are well aligned. We're curious if anyone experienced something similar (plots in screen cap obtained by running camera-imu calibration with `--verbose` option).

example with IMU lags behind camera:
[image]

example with IMU leading camera:
[image]

overall camera predicted angular velocity vs observed IMU gyro signal:
[image]

Additional info:
The IMU we're using is being aggressively LPF'd. Hence the smooth measured IMU signal curve. This introduces ~33ms lag, so with a fixed exposure time of 2ms on stereo global shutter cameras we expected to see a fixed ~31ms lead in camera signal vs IMU signal. Camera-IMU calibration indeed captures this delay reliably. We tried different LPF settings on the IMU - e.g. less aggressive filtering, more noise but less latency and Kalibr picked up these changes quite reliably. However, the variable camera/IMU signal lag is present in every sequence we collected.

ZacharyTaylor, 2018-06-27:
How are you timestamping your data as this could possibly be the issue. If you are timestamping the sensor data when it arrives at the computer you will get unknown and variable delays associated with the time to transfer it plus the delay before computer gets around to processing it.

If your data is timestamped on the sensor, how do you synchronize the clocks between the sensors, as an issue here could also cause the observed behavior.

redbehemoth, 2018-06-27:
Hi Zachary,
Our setup is the following:

1. We have a 2 microcontroller system; stereo cameras on microcontroller-1 and and IMU on microcontroller 2.
2. Stereo cameras are themselves hardware synced. Immediately before exposure starts, the cameras trigger an interrupt on MCU-1. We record this time as camera start of exposure timestamp. We also know the exposure time (fixed to 2ms in the above experiment). 
3. We verified the time sync between MCU-1 and MCU-2 (standard ping-pong sync method) - also checked with oscilloscope and they drift approx 1ms over 90min. We sync between the 2 MCUs every 30min.
4. We are timestamping IMU packets as they arrive on MCU-2. We haven't found a way to trigger a packet read on IMU - it is simply set to transfer its current measurement to MCU-2 at 200hz; each measurement from IMU triggers an interrupt on MCU-2 and we record this as IMU timestamp.

Could there be some problem with the IMU data itself? Do you have some suggestions on experiments we can do to narrow down the source of the variable lag we're observing?

schneith, 2018-06-28:
Are these the plots popping up before the optimization? If so, these are showing the results of the time-shift initialization for the non-linear least squares problem. In this problem the IMU biases are neglected and assumed to be zero (https://github.com/ethz-asl/kalibr/blob/master/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccSensors.py#L215). Maybe this could explain the inaccuracies you are seeing?

redbehemoth, 2018-06-28:
@schneith 
Could IMU bias impact the phase rather than just the amplitude of Omega? Also, these variations are ~30ms to ~45ms occurring on the order seconds. Example: plot 3 shows transition from ahead (at ~3s in the plot) to behind (~5.6s in the plot).  I would expect bias to make a significant impact on the order of several 10s of seconds.

Nadonino, 2018-06-28:
Hi @schneith, thanks for your reply. I work with @redbehemoth  and had a slightly more direct question.
Is looking at the local time offsets between the red graph (IMU gyro signal) and the green graph (camera predicted angular velocity) an accurate way to determine if there is a variable lag between the IMU and the camera?