ethz-asl/kalibr #201: kalibr_camera_validator hangs

rikobrandes, 2018-07-04:
ROS kinetic on Ubuntu 16.04 LTS

I run the kalibr validation, which reads in camera intrinsics and then does nothing. Any ideas?

```
rosrun kalibr kalibr_camera_validator --cam '/home/rnd/git/Knowledge_Base/Calibration/Multicamera to IMU_calibration/calibration_results/B1_frontcam/Bagmerged_left_middle_rightview_checkerboard/camchain-mntnfs_sharecheckerboardB12018-06-29-15_B1_merged.yaml' --target '/home/rnd/git/Knowledge_Base/Calibration/Multicamera to IMU_calibration/calibration_targets/used/checkerboard.yaml' --verbose
```
Shell output
```
Initializing calibration target:
  Type: checkerboard
  Rows
    Count: 13
    Distance: 0.1 [m]
  Cols
    Count: 13
    Distance: 0.1 [m]
initializing camera geometry
Camera /nvidia_camera/port_B1/image_raw:
  Camera model: pinhole
  Focal length: [935.1607151211006, 942.3520842979963]
  Principal point: [927.8464970718626, 632.5681178685664]
  Distortion model: radtan
  Distortion coefficients: [-0.24856508942204, 0.04402551605213663, -0.0012953616934129153, 0.0013295676957414276]
```

melodybinbin, 2018-08-08:
@rikobrandes 
Hello,
I have met the same prolem. Do you solve it? Thanks in advance!

Tim

inecho, 2018-08-21:
i have the same problem

GITSHOHOKU, 2018-09-04:
I have met this problem too.Have you  solved it? Thanks in advance~

rikobrandes, 2018-09-04:
Unfortunately i have not solved it.

Mit freundlichen Grüßen / Best Regards

Riko Brandes

Abteilung Autonome Logistikanwendungen
Department Autonomous Logistics

Mobil:  +49 172 6746 806
Mail:   Riko.Brandes@streetscooter.eu<mailto:Riko.Brandes@streetscooter.eu>
Internet:       www.streetscooter.eu<http://www.streetscooter.eu>

Postanschrift:
StreetScooter GmbH
Jülicher Str. 191
D-52070 Aachen
Germany                 Rechnungsanschrift:
StreetScooter GmbH c/o Deutsche Post AG
Finance & HR Operations Deutschland
Abt. Accounting for Aff. Companies
D-50577 Köln
Germany

[Auszubildender (w/m) zur Industriekauffrau/zum Industriekaufmann]<https://www.streetscooter.eu/wp-content/uploads/2017/06/250_Azubi-Industriekf.-2018.pdf>             Besuchen Sie uns auf der IAA Nutzfahrzeuge - Halle 13 Stand C73
[cid:image84a445.JPG@33992343.499966dd]<https://www.iaa.de/>

STREETSCOOTER is a company of Deutsche Post DHL Group
GOGREEN – Climate Protection with Deutsche Post DHL Group
Please consider your environmental responsibility before printing this E-Mail.
-----------------------------------------------------------------------------------------------------------------------------------------------
Registered office: Aachen  -  Register court: Aachen, HRB 16392
Managing Directors: Prof. Dr.-Ing. Achim Kampker (CEO), Fabian Schmitt (CTO), Sebastian Peter (CFO)
-----------------------------------------------------------------------------------------------------------------------------------------------
The information contained in this email transmission is confidential and may be privileged.
If you are not the intended recipient, any use, dissemination, distribution, publication,
or copying of the information contained in this email is strictly prohibited.
If you have received this email in error, please immediately notify me by calling the above
number and delete the email from your system. Thank you for your co-operation.

PetWorm, 2018-12-19:
I met this problem too on ROS kinetic of 16.04.
is that possible this is because of the platform?

rikobrandes, 2018-12-20:
As far as I remember I solved it using a docker container running Ubuntu 14 and ros indigo.

PetWorm, 2018-12-21:
> As far as I remember I solved it using a docker container running Ubuntu 14 and ros indigo.

thanks

sneheshs, 2019-02-01:
Hello, how did you solve it? Can the Kalibr team provide more info on the validator please?

sneheshs, 2019-02-01:
Ok, I figured out the issues and I have a fix. I will try to do a pull request soon.

GengMingjin2, 2019-03-27:
Hello, any solution for such a problem? I still met the same issue on my machine (Ubuntu 16.04.5 and ROS kinetic). I will be grateful for your help. Thanks!

neorobo, 2019-04-12:
> Ok, I figured out the issues and I have a fix. I will try to do a pull request soon.
What was the issue? Did you submit a PR?

tkkhuu, 2019-04-15:
I'm also interested in this tool in validating my calibration. It also hangs for me at the same point. It opened 2 CV windows and then crashed. Some debugging (by adding print statements) seems that it has something to do with time sync when getting 2 images. I managed to make some progress by making this change https://github.com/ethz-asl/kalibr/issues/33. The call back was able to receive images but then the CV windows crashed as soon as marker was detected.

sneheshs, 2019-04-16:
Hi Guys,

Sorry for not being able to get back to you all. I made a pull request. If you cannot wait for my changes to be merged, you can simply go to my commit in my fork:
https://github.com/sneheshs/kalibr/commit/14f3df108919aa53f48b88beb76e75c78757d3d3

This should be sufficient to address this issue. If you are running the calibration remotely or in docker then there are additional fixes required. Also, depending on your setup you may have to run as root by doing sudo su. Hope this helps.

Best,
Snehesh

neorobo, 2019-04-17:
I'm trying this out on a stereo camera and made your suggested changes in sneheshs@14f3df1, still doesn't seem to work (opens windows for camera images, reports camera geometry for both left and right images in terminal, then hangs). Are you running this on Ubuntu 16.04 and Kinetic?

sneheshs, 2019-04-17:
@neorobo Yes, I am running on Ubuntu 16.04 and ROS Kinetic.

Please share the following:
- command line output
- python version
- which stereo camera and image format does the camera output

We may have to insert a few debugging lines in the code to see where it is failing. Please share the above info and let's see.

Pedrous, 2019-05-14:
> Hi Guys,
> 
> Sorry for not being able to get back to you all.
> 
> First, you should run the kalibr_camera_validator instead of via rosrun, run via python (I used python3 but I have tried python2 and that also works)
> `python3 kalibr_camera_validator --cam camchain.yaml --target target.yaml `
> Also, the person who created the issue has a really long path with spaces, I would avoid that. Try simplifying the path and see if that works.
> 
> For multi-cam, the issue was not getting perfectly time synchronized images so I implemented approximate time synchronizer just like they have done on the actual calibration module.
> 
> I made a pull request. If you cannot wait for my changes to be merged, you can simply go to my commit in my fork: [sneheshs@14f3df1](https://github.com/sneheshs/kalibr/commit/14f3df108919aa53f48b88beb76e75c78757d3d3)
> `https://github.com/sneheshs/kalibr/commit/14f3df108919aa53f48b88beb76e75c78757d3d3 `
> 
> This should be sufficient to address this issue. If you are running the calibration remotely or in docker then there are additional fixes required. Also, depending on your setup you may have to run as root by doing sudo su. Hope this helps.
> 
> Best,
> Snehesh

Thank you sneheshs, this worked out perfectly and I can confirm that I was able to get it functional in docker as well.

If you use docker you, you just need to share your desktop by 
sudo docker run -it --net=host --env="DISPLAY" --volume="$HOME/.Xauthority:/root/.Xauthority:rw"

and also you need to have 'x11-apps' installed in your docker container. But the visualization will be very though...

IrisGreen, 2019-05-20:
Hi guys,
I run as @sneheshs said ,but it says as follow:
No module named aslam_cv 
I have  already catkin build the kalibr_workspace and don’t know what should I do. Any help would be appreciate!
BTW, should I open another two terminal to run roscore and play a image-topic rosbag?

sneheshs, 2019-05-20:
> Hi guys,
> I run as @sneheshs said ,but it says as follow:
> No module named aslam_cv
> I have already catkin build the kalibr_workspace and don’t know what should I do. Any help would be appreciate!
> BTW, should I open another two terminal to run roscore and play a image-topic rosbag?

Can you provide more info which step you failed, what cameras are you using, which ubuntu are you using, command line output, python version etc?

IrisGreen, 2019-05-23:
> Can you provide more info which step you failed, what cameras are you using, which ubuntu are you using, command line output, python version etc?
Hi, @sneheshs ,sorry for the late reply.
I use a stereo camera MYNTEYE S2100,Ubuntu 16.04,python3(have tried python2.7)
I opened a terminal in
`~/kalibr_workspace/arc/Kalibr/aslam_offline_calibration/kalibr/python`
And the command line
`python3 kalibr_camera_validator —cam camchain.yaml —target april.yaml`
Then the error showed
`No module named aslam_cv`
BTW,I have tried to use a command in terminal 1
`roscore`
In terminal 2
`kalibr_camera_validator —cam camchain.yaml —target april.yaml`
In terminal3
`rosbag play omg.bag -r 4.0 /left_image /right_image`
Then the error showed
`Conversion is only valid for 1 or 2 dimensions. Argument has 3 dimensions`

EmiyaEstelle, 2019-10-10:
Hi @sneheshs 
I seem to meet the same problem about the kalibr_camera_validator

EmiyaEstelle, 2019-10-10:
Hi @sneheshs
I seem to meet the same problem about the kalibr_camera_validator.
I see the commits you said above but it may not work.
First, I try to use python (Ubuntu 16.04 and python 2) but it shows me a error
[image]

Second I try to use ros. In one terminal I run the roscore. And in another, I run the validator. But it will be freeze 
[image]
I am not familiar with python and I can not find out the reason.

[image]

HwangDaeHyun, 2020-07-27:
I solved this problem with comment 'cv2.namedWindow(self.windowName, 1)' line in 'def __init__' of MonoCameraValidatior class.

[image]

mintar, 2022-05-03:
This should be fixed once #515 is merged.

commit 71311022 (2022-05-03): camera_validator: Fix thread getting stuck (empty windows)

Moved the creation of the cv2.namedWindow from the MonoCameraValidator
constructor to the callback function. Otherwise, this caused the
following error:

    QObject::startTimer: Timers cannot be started from another thread

... with the result that the image windows remained black and the node
became stuck.

Fixes #201 .
closed 2022-05-03 by goldbattle