ethz-asl/kalibr #202: ImportError: cannot import name NavigationToolbar2Wx , Ubuntu 16.04

salahkhan94, 2018-07-09:
Kalibr got compiled perfectly, I tried running the Rosbag datasets I downloaded from the wiki page I got the following error:

```
sal@sal:~/Downloads/static$ kalibr_calibrate_cameras --target april_6x6.yaml --bag static.bag --models pinhole-equi pinhole-equi omni-radtan omni-radtan --topics /cam0/image_raw /cam1/image_raw /cam2/image_raw /cam3/image_raw
importing libraries
Traceback (most recent call last):
  File "/home/sal/kalibr_workspace/devel/bin/kalibr_calibrate_cameras", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/sal/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_cameras", line 7, in <module>
    from sm import PlotCollection
  File "/home/sal/kalibr_workspace/src/Kalibr/Schweizer-Messer/sm_python/python/sm/PlotCollection.py", line 8, in <module>
    from matplotlib.backends.backend_wxagg import NavigationToolbar2Wx as Toolbar
ImportError: cannot import name NavigationToolbar2Wx

```

Am I missing any dependencies?

jasonir129, 2018-11-09:
thanks for figuring this out!

bhagirathioctal, 2019-11-16:
Hi someone had had an error related to PlotCollection?  

`~/kalibr_ws/src/kalibr/aslam_offline_calibration/kalibr/data$ ../python/kalibr_calibrate_cameras --target april_6x6.yaml --bag recor
d_at_4fps.bag --models pinhole-equi pinhole-equi pinhole-equi --topics /cam0/image_raw /cam1/image_raw /cam2/image_raw /cam3/image_raw
importing libraries
Traceback (most recent call last):
  File "../python/kalibr_calibrate_cameras", line 7, in <module>
    from sm import PlotCollection
ImportError: cannot import name PlotCollection
`

mahammadirfan, 2021-09-24:
@bhagirathioctal Hey have you found the solution? I am getting the same error ?