ethz-asl/kalibr #209: calibration file for Optor

tiralonghipol, 2018-08-02:
Hi. I would like to know if anyone has tried to calibrate this sensor:
https://www.robotshop.com/jp/en/optor-visual-inertial-camera.html

I'm having troubles with the camera/imu synchronization

Thanks

Blackshaddock, 2018-08-29:
I also use optor camera. did you run https://github.com/HKUST-Aerial-Robotics/VINS-Mono?

Blackshaddock, 2018-08-29:
@tiralonghipol I also use optor camera. did you use it run https://github.com/HKUST-Aerial-Robotics/VINS-Mono?

tiralonghipol, 2018-08-29:
@Blackshaddock  I'm actually conducting many experiments, VINS was one of them. The problem with the optor is the different arrival times between cam0 and cam1 and no hardware triggering, as you understand from this discussion 
https://github.com/ethz-asl/rovio/issues/192

If you have any suggestion on how to modify the sdk in order to get it work, let me know

tiralonghipol, 2018-10-04:
Hi.
After many tries I assumed that this device is not hardware synchronized as declared
here you can read more in deep what the problem is: ethz-asl/rovio#192

closed 2022-05-04 by goldbattle