ethz-asl/kalibr #211: I made a tutorial video for beginners

indigomega, 2018-08-12:
Hey there, as part of my series on YouTube I made a basic tutorial about using Kalibr. I didn’t really know where else to put this so I marked it as a problem (close it when you wish). There are some unrelated comments in the beginning but the rest of the video talks about Kalibr. Please give it a look and if you think it’s good enough I hope you could add it to the wiki. Thanks!

https://m.youtube.com/watch?v=puNXsnrYWTY

schneith, 2018-08-13:
That's awesome, thanks for sharing your video. I'm sure that this will be very helpful for other trying to use kalibr. I'll add it to the wiki pages.

One small comment about the calibration target: the calibration is quite sensitive to the flatness of target. you might improve the results by using a more flatter target.

schneith, 2018-08-13:
The video is now linked on the main wiki page. Do you think it makes sense to add it to another section?

indigomega, 2018-08-15:
Great stuff! Closing now

closed 2018-08-15 by indigomega