ethz-asl/kalibr #212: Build errors with Ubuntu 16.04, Cmake 3.9 and boost 1.67

Kaju-Bubanja, 2018-08-16:
I'm trying to build kalibr using catkin build, but I get following error: 

```In file included from /kalibr/aslam_cv/aslam_time/src/duration.cpp:2:0:
/kalibr/aslam_cv/aslam_time/include/aslam/implementation/Duration.hpp: In instantiation of ‘boost::posix_time::time_duration aslam::DurationBase<T>::toBoost() const [with T = aslam::Duration]’:
/kalibr/aslam_cv/aslam_time/src/duration.cpp:36:16:   required from here
/kalibr/aslam_cv/aslam_time/include/aslam/implementation/Duration.hpp:168:27: error: no matching function for call to ‘boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000000l>::subsecond_duration(double)’
   return bt::seconds(sec) + bt::microseconds(nsec / 1000.0);
```

I'm kinda confused by the error message because the function exists, but I guess it is typed with different input types? I think the easiest way to resolve is to change my boost, cmake, gcc version to match the version that was tested with kalibr and that is known to build.

NikolausDemmel, 2018-08-16:
I had this error the other day compiling some ros time libraries on mac. I think it should be `nsec / 1000` (to not convert to double).

Kaju-Bubanja, 2018-08-16:
Thanks for the hint, looking at the code some more this was also my guess. Do you have any idea why this builds for some people and not for others? I guess it's a different compiler version? Or another version of something in the toolchain. I would like to know the cause so I can just switch whatever version it is that is wrong on my pc, because I don't feel like going around fixing all these small little things, because after fixing this I got the following error and I suppose there are more to come: 
```
/kalibr/aslam_optimizer/aslam_backend_expressions/src/MatrixTransformation.cpp:20:62: error: no 
match for ‘operator==’ (operand types are ‘Eigen::internal::enable_if<true, 
Eigen::IndexedView<Eigen::Matrix<double, 3, 3>, Eigen::internal::SingleRange, double> >::type {aka 
Eigen::IndexedView<Eigen::Matrix<double, 3, 3>, Eigen::internal::SingleRange, double>}’ and ‘int’)
       if (_UpdatePattern(i%3,floor(static_cast<double>(i/3)))==1){
```

NikolausDemmel, 2018-08-16:
btw the first error seems to be caused by a change in boost: https://github.com/mikepurvis/ros-install-osx/issues/116#issuecomment-386189047

Kaju-Bubanja, 2018-08-16:
Thank you for the lead, I will try to build it with 1.63 later and see if that fixes the issues.

ffurrer, 2018-08-16:
Yup, some other people fixed the same issue by changing types from double to long:
https://gitlab.com/orcus/orcus/merge_requests/29/diffs

Would be nice if you could open a PR with these fixes, if you succeed :).

Kaju-Bubanja, 2018-08-16:
Could fix the issue, PR is here: https://github.com/ethz-asl/kalibr/pull/213

closed 2018-08-16 by Kaju-Bubanja