ethz-asl/kalibr #214: kalibr_camera_focus Problems with image format

kauevestena, 2018-08-17:
After call the script with this command, with a logitech c-270 webcam with 480 x 640 of resolution:

        `kalibr_camera_focus` --topic `/camera/image_raw`

[image]

I just receive the error message that can be seen at the print screen, that is:
> Assertion failed (type == CV_32FC1 || type == CV_32FC2 || type == CV_64FC1 || type == CV_64FC2) in dft

so I opened the code at the `dft` call, and asked for the array sizes, and it outputs `921600` wich is the result of the multiplication `480*640*3` so the script is trying to pass a 3-Band Array to input for the `dft` function.

Checking the documentation here:

> https://docs.opencv.org/3.0-beta/modules/core/doc/operations_on_arrays.html?highlight=dft#cv2.dft

it says that with the `cv2.DFT_COMPLEX_OUTPUT` flag, the function will

>performs a forward transformation of 1D or 2D real array...

AntaresZhang, 2018-12-10:
This also solved my problem, Thanks very much!

goldbattle, 2022-06-02:
Should be fixed. Reopen if this is still an issue for the ROS tools.

closed 2022-06-02 by goldbattle