ethz-asl/kalibr #215: Paper to do the multiple camera calibration

WillenPeng, 2018-08-23:
HI, 

Can I ask which paper do you follow to do the multiple camera calibration?

Thanks

SSjn, 2019-08-18:
This is also my wondering about. Can somebody point it out, please?

NikolausDemmel, 2019-09-24:
Maybe check the ones in the wiki, e.g. IROS 2013: https://github.com/ethz-asl/kalibr#references

NikolausDemmel, 2019-09-24:
Ah you are right, most are about the cam-imu calibration. 

The most relevant out of these is then probably the last one (https://github.com/ethz-asl/kalibr/wiki/multiple-camera-calibration#references) that seems to describe the batch optimization and in particular the information selection strategy.

For the other parts of camera calibration, I think these principles are very old, so they are not presented in these publications since this is no novelty in Kalibr. You could try to check the papers on the different camera models, on AprilGrids or the other pattern types, and see if you can find what you are looking for in there or in their references.

SSjn, 2019-09-25:
Thank you for your detailed answer. I'll continue to look for it and then feed back the progress here.

dongshengLiu, 2020-10-12:
> Thank you for your detailed answer. I'll continue to look for it and then feed back the progress here.

I want to separate the multi-camera calibration part without relying on ROS. Do you have a recommendation for the paper on Kalibr multi-camera calibration?

closed 2022-05-04 by goldbattle

Jianan92, 2022-05-04:
您好，您的邮件已收到！