ethz-asl/kalibr #216: wrong result of the calibration of camera and imu.

ttomchy, 2018-08-25:
hello,Thanks for providing the algorithm for calibrate the camera ,I am trying to calibrate one camera (pointgrey camera resolution is: 2448*2048) and one IMU (xsens MTI-30-2A5G4).  But I the translation result is apparently wrong. 
I am using the code (https://github.com/rpng/kalibr_allan） to get the imu parameter ,I recorded  about 4 hours' imu data . The following figures shows the imu parameter.
[image]
I use ros message filters (approximate_time) to get synchronized camera and imu topics. I have tried the following code to calibrate the camera and imu.
`kalibr_calibrate_imu_camera --target checkerboard_10x9_45x45cm.yaml  --cam camchain.yaml --imu imu_adis16448.yaml --bag test.bag --show-extraction --time-calibration --bag-from-to 20 140`
Here is the output `importing libraries
Initializing IMUs:
  Update rate: 7
  Accelerometer:
    Noise density: 0.07120086 
    Noise density (discrete): 0.188379768694 
    Random walk: 0.00069865
  Gyroscope:
    Noise density: 0.05389505
    Noise density (discrete): 0.142592899197 
    Random walk: 0.00013758
Initializing imu rosbag dataset reader:
	Dataset:          test.bag
	Topic:            /imu/data
bagstart 1535175681.16
baglength 157.035122156
[ WARN] [1535203375.243458]: BagImuDatasetReader: truncated 261 / 1101 messages.
	Number of messages: 1101
Reading IMU data (/imu/data)
  Read 840 imu readings over 119.8 seconds                   
Initializing calibration target:
  Type: checkerboard
  Rows
    Count: 9
    Distance: 0.045 [m]
  Cols
    Count: 10
    Distance: 0.045 [m]
Initializing camera chain:
Camera chain - cam0:
  Camera model: pinhole
  Focal length: [1817.92373, 1815.088909]
  Principal point: [1233.884, 1018.055]
  Distortion model: equidistant
  Distortion coefficients: [-0.329746, 0.127861, 0.000286, -0.000362]
  baseline: no data available
Initializing camera rosbag dataset reader:
	Dataset:          test.bag
	Topic:            /camera/image_mono
[ WARN] [1535203386.120824]: BagImageDatasetReader: truncated 261 / 1101 images.
	Number of images: 1101
Extracting calibration target corners
  Extracted corners for 823 images (of 840 images)                              

Building the problem
	Spline order: 6
	Pose knots per second: 100
	Do pose motion regularization: False
		xddot translation variance: 1000000.000000
		xddot rotation variance: 100000.000000
	Bias knots per second: 50
	Do bias motion regularization: True
	Blake-Zisserman on reprojection errors -1
	Acceleration Huber width (sigma): -1.000000
	Gyroscope Huber width (sigma): -1.000000
	Do time calibration: True
	Max iterations: 30
	Time offset padding: 0.010000
Estimating time shift camera to imu:

Initializing a pose spline with 11977 knots (100.000000 knots per second over 119.767592 seconds)
  Time shift camera to imu (t_imu = t_cam + shift):
0.0

Estimating imu-camera rotation prior
[... 289 lines cut ...]
      [[ 0.00154999 -0.00001688  0.00093119]
       [-0.00637655 -0.0051548  -0.00255806]
       [-0.00015744 -0.00028733 -0.00027278]]
    C_gyro_i:
      [[ 0.99999771 -0.00169169  0.00131341]
       [ 0.00169353  0.99999759 -0.00139933]
       [-0.00131104  0.00140155  0.99999816]]
  Accelerometer: 
    M:
      [[ 1.00028102  0.          0.        ]
       [-0.00025133  1.00224152  0.        ]
       [ 0.00036224  0.00199759  1.00057533]]

  Saving camera chain calibration to file: camchain-imucam-test.yaml

  Saving imu calibration to file: imu-test.yaml
  Detailed results written to file: results-imucam-test.txt
Generating result report...
/usr/lib/pymodules/python2.7/matplotlib/collections.py:548: FutureWarning: elementwise comparison failed; returning scalar instead, but in the future will perform elementwise comparison
  if self._edgecolors == 'face':
  Report written to report-imucam-test.pdf

`

However It seems that the result is almost the same (both are wrong result.) is anyone can help me ?

RuanJiaJun, 2021-12-16:
I also encountered this problem. Have you solved it now? Can you share the solution? Thank you！

goldbattle, 2022-06-02:
Closing due to age.

closed 2022-06-02 by goldbattle