ethz-asl/kalibr #223: Problem with single imu and single cam  Optimization failed

yangtzey, 2018-09-05:
Terminal output：
kalibr_calibrate_imu_camera --bag awsome.bag --cam camchain.yaml --imu imu.yaml --target target.yaml --show-extraction
importing libraries
Initializing IMUs:
  Update rate: 100
  Accelerometer:
    Noise density: 6.46658958941e-12 
    Noise density (discrete): 6.46658958941e-11 
    Random walk: 9.0201496474e-15
  Gyroscope:
    Noise density: 0.0021916404834
    Noise density (discrete): 0.021916404834 
    Random walk: 2.67503649373e-05
Initializing imu rosbag dataset reader:
	Dataset:          awsome.bag
	Topic:            /imu
	Number of messages: 13288
Reading IMU data (/imu)
  Read 13288 imu readings over 137.1 seconds                         
Initializing calibration target:
  Type: checkerboard
  Rows
    Count: 3
    Distance: 0.028778 [m]
  Cols
    Count: 3
    Distance: 0.028778 [m]
Initializing camera chain:
Camera chain - cam0:
  Camera model: pinhole
  Focal length: [672.937487, 671.829359]
  Principal point: [315.128219, 224.870491]
  Distortion model: radtan
  Distortion coefficients: [-0.05919, -0.062466, -0.003951, -0.002649]
  baseline: no data available
Initializing camera rosbag dataset reader:
	Dataset:          awsome.bag
	Topic:            /usb_cam/image_raw
	Number of images: 4098
GLib-GIO-Message: Using the 'memory' GSettings backend.  Your settings will not be saved or shared with other applications.
Extracting calibration target corners
  Extracted corners for 1413 images (of 4098 images)                              

Building the problem
	Spline order: 6
	Pose knots per second: 100
	Do pose motion regularization: False
		xddot translation variance: 1000000.000000
		xddot rotation variance: 100000.000000
	Bias knots per second: 50
	Do bias motion regularization: True
	Blake-Zisserman on reprojection errors -1
	Acceleration Huber width (sigma): -1.000000
	Gyroscope Huber width (sigma): -1.000000
	Do time calibration: False
	Max iterations: 30
	Time offset padding: 0.010000

Estimating imu-camera rotation prior

Initializing a pose spline with 13701 knots (100.000000 knots per second over 137.005007 seconds)
Gravity was intialized to [-2.36396971  0.38156876  9.50970428] [m/s^2]
  Orientation prior camera-imu found as: (T_i_c)
[[ 0.99529898  0.00966187 -0.09636692]
 [-0.00714784  0.99962592  0.02639933]
 [ 0.09658594 -0.02558641  0.99499572]]
  Gyro bias prior found as: (b_gyro)
[ 0.12630099  0.00248894 -0.03040858]

Initializing a pose spline with 13705 knots (100.000000 knots per second over 137.045007 seconds)

Initializing the bias splines with 6852 knots

Adding camera error terms (/usb_cam/image_raw)
  Added 1413 camera error terms                               

Adding accelerometer error terms (/imu)
  Added 13284 of 13288 accelerometer error terms (skipped 4 out-of-bounds measurements)

Adding gyroscope error terms (/imu)
  Added 13284 of 13288 gyroscope error terms (skipped 4 out-of-bounds measurements)

Before Optimization
===================
Normalized Residuals
----------------------------
Reprojection error (cam0):     mean 65632075674.2, median 1.83902475312, std: 7.35268713331e+12
Gyroscope error (imu0):        mean 199.34385127, median 78.8340659789, std: 265.711001816
Accelerometer error (imu0):    mean 1.2241773027e+11, median 86200187256.7, std: 1.39298334649e+11

Residuals
----------------------------
Reprojection error (cam0) [px]:     mean 65632075674.2, median 1.83902475312, std: 7.35268713331e+12
Gyroscope error (imu0) [rad/s]:     mean 4.3689005456, median 1.7277593047, std: 5.82342988464
Accelerometer error (imu0) [m/s^2]: mean 7.91625220123, median 5.5742123352, std: 9.00785160663

Optimizing...
Using the block_cholesky linear system solver
Using the levenberg_marquardt trust region policy
Using the block_cholesky linear system solver
Using the levenberg_marquardt trust region policy
Initializing
Optimization problem initialized with 27427 design variables and 39131 error terms
The Jacobian matrix is 104828 x 123410
[0.0]: J: 6.79584e+29
**CHOLMOD warning: matrix not positive definite
[WARNING] System solution failed
CHOLMOD warning: matrix not positive definite
[WARNING] System solution failed
CHOLMOD warning: matrix not positive definite
[WARNING] System solution failed
CHOLMOD warning: matrix not positive definite
[WARNING] System solution failed
CHOLMOD warning: matrix not positive definite
[WARNING] System solution failed
CHOLMOD warning: matrix not positive definite
[WARNING] System solution failed
CHOLMOD warning: matrix not positive definite
[WARNING] System solution failed**
Last step was a regression. Reverting
[1]: J: 5.25068e+32, dJ: -5.24389e+32, deltaX: 0.510157, LM - lambda:3.43597e+11 mu:256
[2]: J: 1.10111e+29, dJ: 5.69473e+29, deltaX: 0.00251212, LM - lambda:1.75922e+14 mu:512
[3]: J: 1.8087e+28, dJ: 9.2024e+28, deltaX: 0.00411214, LM - lambda:1.21332e+14 mu:2
[4]: J: 3.18544e+27, dJ: 1.49015e+28, deltaX: 0.00572804, LM - lambda:8.36216e+13 mu:2
[5]: J: 7.49866e+26, dJ: 2.43558e+27, deltaX: 0.0065875, LM - lambda:5.73953e+13 mu:2
[6]: J: 3.35137e+26, dJ: 4.14729e+26, deltaX: 0.00598049, LM - lambda:3.86541e+13 mu:2
[7]: J: 2.42019e+26, dJ: 9.31176e+25, deltaX: 0.0101304, LM - lambda:2.34986e+13 mu:2
[8]: J: 1.55361e+26, dJ: 8.66587e+25, deltaX: 0.052003, LM - lambda:7.83288e+12 mu:2
[9]: J: 5.24328e+25, dJ: 1.02928e+26, deltaX: 0.180743, LM - lambda:2.61096e+12 mu:2
[10]: J: 4.36856e+24, dJ: 4.80643e+25, deltaX: 0.223524, LM - lambda:8.7032e+11 mu:2
[11]: J: 1.88674e+23, dJ: 4.17989e+24, deltaX: 0.105611, LM - lambda:2.90107e+11 mu:2
[12]: J: 1.6031e+22, dJ: 1.72643e+23, deltaX: 0.08933, LM - lambda:9.67022e+10 mu:2
[13]: J: 5.17342e+21, dJ: 1.08576e+22, deltaX: 0.110582, LM - lambda:3.37249e+10 mu:2
[14]: J: 1.05187e+21, dJ: 4.12155e+21, deltaX: 0.0475756, LM - lambda:3.21336e+10 mu:2
^C^C[15]: J: 2.24171e+20, dJ: 8.27699e+20, deltaX: 0.0497797, LM - lambda:2.46266e+10 mu:2
[16]: J: 5.16843e+19, dJ: 1.72487e+20, deltaX: 0.0328523, LM - lambda:1.66466e+10 mu:2
[17]: J: 1.17887e+19, dJ: 3.98956e+19, deltaX: 0.0478505, LM - lambda:7.85809e+09 mu:2
[18]: J: 2.29491e+18, dJ: 9.49376e+18, deltaX: 0.0561758, LM - lambda:2.61936e+09 mu:2
[19]: J: 4.4464e+17, dJ: 1.85027e+18, deltaX: 0.0534144, LM - lambda:1.25737e+09 mu:2
[20]: J: 1.09473e+17, dJ: 3.35168e+17, deltaX: 0.0499163, LM - lambda:6.72719e+08 mu:2
[21]: J: 2.78428e+16, dJ: 8.16298e+16, deltaX: 0.0385642, LM - lambda:4.68972e+08 mu:2
[22]: J: 9.53238e+15, dJ: 1.83105e+16, deltaX: 0.0328745, LM - lambda:3.09196e+08 mu:2
[23]: J: 2.56085e+15, dJ: 6.97153e+15, deltaX: 0.0224583, LM - lambda:2.59689e+08 mu:2
Last step was a regression. Reverting
[24]: J: 9.75209e+15, dJ: -7.19124e+15, deltaX: 0.0335488, LM - lambda:1.4072e+08 mu:2
[25]: J: 1.15744e+15, dJ: 1.40341e+15, deltaX: 0.00482438, LM - lambda:5.62879e+08 mu:4
Last step was a regression. Reverting
[26]: J: 2.35069e+15, dJ: -1.19325e+15, deltaX: 0.0228396, LM - lambda:1.87626e+08 mu:2
[27]: J: 1.07579e+15, dJ: 8.16508e+13, deltaX: 0.00198407, LM - lambda:7.50505e+08 mu:4
[28]: J: 9.61807e+14, dJ: 1.13983e+14, deltaX: 0.0144416, LM - lambda:2.50168e+08 mu:2
[29]: J: 5.88816e+14, dJ: 3.72991e+14, deltaX: 0.0104014, LM - lambda:2.50176e+08 mu:2
Last step was a regression. Reverting
[30]: J: 1.22271e+16, dJ: -1.16383e+16, deltaX: 0.0364705, LM - lambda:8.3392e+07 mu:2

[ WARN] [1536131146.902323]: Shutting down! (CTRL+C)
[ERROR] [1536131146.932736]: Optimization failed!
Traceback (most recent call last):
  File "/home/yangtze/work/Code/kalibr/devel/bin/kalibr_calibrate_imu_camera", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/yangtze/work/Code/kalibr/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera", line 236, in <module>
    main()
  File "/home/yangtze/work/Code/kalibr/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera", line 206, in main
    iCal.optimize(maxIterations=parsed.max_iter, recoverCov=parsed.recover_cov)
  File "/home/yangtze/work/Code/kalibr/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccCalibrator.py", line 179, in optimize
    raise RuntimeError("Optimization failed!")
RuntimeError: Optimization failed!

could you help me？
what is this mean?
[WARNING] System solution failed
CHOLMOD warning: matrix not positive definite it mean
Why did it happen？
RuntimeError: Optimization failed!

schneith, 2018-09-05:
Often this indicates that something is wrong with the input data. Did you have a look at the accelerometer / gyro measurements? Do they look good (units, vibrations, shocks, right-hand system,...) ?

yangtzey, 2018-09-06:
Hi Schneith, thank you very much!
my device is ADXL345， the z-axis is reversed. after changed the direction.there was no problem that"**CHOLMOD warning: matrix not positive definite
[WARNING] System solution failed**"
one more question:How to determine the angular velocity direction of the three axis?

schneith, 2018-09-06:
You can verify the directions by rotating the IMU by hand if you know the approximate coordinate frame of the gyroscope (you can probably find this in the datasheet). You then rotate e.g. around the x-axis and you should get a positive output if you rotate like this:
https://en.wikipedia.org/wiki/Right-hand_rule#/media/File:Right-hand_grip_rule.svg

Then repeat this for all axis and flip signs if necessary.

yangtzey, 2018-09-07:
@schneith ,thanks again！

closed 2018-09-07 by yangtzey

whxtsangle, 2021-04-26:
I need some help.Is there a relationship between the camera coordinate system and the inertial coordinate system

mhyoosefian, 2021-11-18:
> I need some help.Is there a relationship between the camera coordinate system and the inertial coordinate system

Hi,

What do you exactly mean by relationship? One primary use of Kalibr is to find the rotation matrix and the translation vector between IMU and camera coordinate frames, i.e. `T_ci`.

Regards