ethz-asl/kalibr #226: Generating kalibr calibration for Gazebo wide-angle cameras

skohlbr, 2018-09-21:
Since some time, Gazebo provides [support for the simulation of wide-angle cameras](http://gazebosim.org/tutorials?tut=wide_angle_camera&branch=wideanglecamera). The mapping from the model used within Gazebo and the ones available in kalibr appears non-trivial at first sight (mileage might differ depending on level of expertise ;) ). I'd be interested in information if somebody already took a look at this and/or any hints on how to perform the parameter mapping.

NikolausDemmel, 2018-09-21:
I haven't looked at Gazebo's wide-angle camera before, but from the wiki it looks it supports a range of different projection functions. Question is: What is your aim with this? Do you have a fixed parameterization in Gazebo and would like a suitable model for Kalibr? Or are you free to choose the parameters in Gazebo?

The following models in Kalibr are suitable for fisheye cameras (see also https://github.com/ethz-asl/kalibr/wiki/supported-models):
 - pinhole-fov
 - pinhole-equi
 - omni-radtan
 - eucm-none
 - ds-none

Our recent publication (https://vision.in.tum.de/research/vslam/double-sphere) summarizes all of them (except for the 'radtan' part of omni-radtan; the paper describes only what would be called 'omni-none').

Looks like in gazebo you have fine-control over the mapping, in particular with the `custom` lens type, but all are radially symmetric (all of the Kalibr models above are radially symmetric, except 'omni-radtan' which can handle some tangential distortion).

Another good resource is the original publication for the Kannala-Brandt model (http://www.ee.oulu.fi/mvg/files/pdf/pdf_697.pdf), which explains that all these different projection type (Stereographic, Equidistant, Equisolid, Orthographic, ...) are handled by approximating the mapping with a polynomial. A simplified version of the KB model is available in Kalibr as 'pinhole-equi' (4 polynomial coefficients, radially-symmetric), so I expect this to handle a range of parameters in Gazebo well.

As for the other models, I don't really know for which variants they work well. We only tested for a range of real world lenses and found they all work ok. The main advantage of eucm and ds is that they have a good trade-off of accuracy and speed (e.g. for real-time, optimization-based slam). fov is a bit limited (only 1 parameter). Omni-radtan is flexible (I guess) and in particular suitable if you also have small tangential distortions.

So my recommendation would be to start with pinhole-equi, and maybe try others as well. A parameter mapping can probably be deduced for specific cases (e.g. for the "equidistant" mapping it looks like you can probably deduce the coefficients, with higher-order being 0). But otherwise the best is probably to just simulate some calibration images and run Kalibr.

One caveat for Kalibr's implementation of pinhole-equi and pinhole-fov: While the KB and FOV models can in principle handle FOV > 180 deg, since they are implemented in Kalibr as pinhole with distortion, in our experience they stop to perform well as FOV approaches 180 deg.

skohlbr, 2018-09-25:
Thanks for the answer! We're using kalibr-calibrated 360 degree cameras on our real robots with great results (generating virtual pinhole projections, using them for object detection, point cloud coloring..), but they're missing from our otherwise comprehensive simulation setup so far. So what I'm interested in is a setup that roughly provides the same resolution and FOV, but otherwise choice of parameters is basically free, as long as a kalibr calibration (that might be different from the one used on the real robot) allows reproducing similar artifacts as on the real robot.

skohlbr, 2018-09-25:
Yes exactly, calibrated by rigidly mounting two 360 degree cams 90 deg offset from each other (for a total of 4 overlapping fisheye images).

NikolausDemmel, 2018-09-25:
Just to clarify, since I'm not quite sure what you mean by "360 degree camera". Are you talking about a camera that consists of 2x ~180deg FoV fisheye cameras in opposing directions, and since you have 2 of those, you have a total of 4 independent fisheye images, each calibrated with the `omni-radtan` model.

Now you would like a similar setup in Gazebo. You mentioned the exact models don't matter, but I guess you would like again something like 4x around 180 deg FoV, too keep it similar?

So essentially you are asking for a suitable Gazebo model for a simulated ~180 FoV camera, which can be calibrated with Kalibr, ideally with `omni-radtan` to keep things simple, right?

skohlbr, 2018-09-25:
We noticed we require 2 cams (2 x 2 fisheye images) for calibration, as the overlap between the two fisheye images of a single camera was not enough to allow calibration. For actual on-robot use, only a single 360 deg cam (2 x fisheye) is used.

Your last sentence sums up what I want to do nicely. Ofc, it would be possible to make a small script spawning/deleting the calibration board in different locations and thus recording a dataset inside Gazebo, afterwards running calibration (just as with a real cam), but it would be great to just generate kalibr (ideally perhaps 'omni-radtan') params directly from the gazebo wideangle camera params.

NikolausDemmel, 2018-09-26:
Hm... It might be that one of the variants corresponds to the universal camera model (first projecting on a sphere, then onto a pinhole camera with off-set center of projection) such that you compute the ideal parameters for omni-radtan directly (with 0 for distortion parameters), but without closer investigation it's not obvious to me which one.

skohlbr, 2018-10-01:
Ok, so I took the plunge and just implemented a small tool that spawns targets in Gazebo and automatically generates a calibration dataset. What is fairly interesting is that generating a calibration with `omni-radtan` fails with pnp errors, which surprises me, as the data should be close to perfect. Conversely, calibration with `ds-none` works and the estimated camera extrinsics are very close to ground truth.

goldbattle, 2022-05-04:
Track via #231

closed 2022-05-04 by goldbattle