ethz-asl/kalibr #232: Tried to add second view to a given cameraId & timestamp. Maybe try to reduce the approximate syncing tolerance..

xukuanHIT, 2018-10-15:
Hi
    Thankyou for your work! But when I calibrate stereo, an error occered:
" Tried to add second view to a given cameraId & timestamp. Maybe try to reduce the approximate syncing tolerance.."
    What should I do?

maokp, 2019-05-16:
How can I resolve the error?

Should I litmit the frame rate to 5Hz?

alwynmathew, 2022-03-22:
@NaveenSehgal If I have a 15FPS camera what would be the ideal sync tolerance?

l4es, 2024-10-26:
Dear @NaveenSehgal,

I am facing the same error message. I would like to know if it can be ignored or I should throttle the camera topics to 5Hz ? In case the sync tolerance needs to be adjusted, where can I do this, please ?

Thanks in advance and best regards,
Khang