ethz-asl/kalibr #233: Errors when building from source and trying to use the CDE package (Ubuntu 16.04, ROS Kinetic)

NaveenSehgal, 2018-10-16:
Hello, 

I am having trouble using Kalibr on my computer. I have tried both building from source and using the CDE package, but get errors either way. Anyone have any ideas? It seems to be related to Boost and Python. 

Thanks.

#### Building from Source Errors
```
Errors     << numpy_eigen:make /home/naveen/kalibr_workspace/logs/numpy_eigen/build.make.029.log
In file included from /usr/include/numpy/ndarraytypes.h:1777:0,
                 from /usr/include/numpy/ndarrayobject.h:18,
                 from /usr/include/numpy/arrayobject.h:4,
                 from /home/naveen/kalibr_workspace/src/Kalibr/Schweizer-Messer/numpy_eigen/include/numpy_eigen/NumpyEigenConverter.hpp:20,
                 from /home/naveen/kalibr_workspace/src/Kalibr/Schweizer-Messer/numpy_eigen/src/autogen_module/numpy_eigen_export_module.cpp:2:
/usr/include/numpy/npy_1_7_deprecated_api.h:15:2: warning: #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
 #warning "Using deprecated NumPy API, disable it by " \
  ^
In file included from /usr/include/signal.h:316:0,
                 from /usr/include/numpy/npy_interrupt.h:84,
                 from /usr/include/numpy/arrayobject.h:5,
                 from /home/naveen/kalibr_workspace/src/Kalibr/Schweizer-Messer/numpy_eigen/include/numpy_eigen/NumpyEigenConverter.hpp:20,
                 from /home/naveen/kalibr_workspace/src/Kalibr/Schweizer-Messer/numpy_eigen/src/autogen_module/numpy_eigen_export_module.cpp:2:
/home/naveen/kalibr_workspace/src/Kalibr/Schweizer-Messer/numpy_eigen/src/autogen_module/numpy_eigen_export_module.cpp: In function ‘void init_module_libnumpy_eigen()’:
/home/naveen/kalibr_workspace/src/Kalibr/Schweizer-Messer/numpy_eigen/src/autogen_module/numpy_eigen_export_module.cpp:257:2: error: return-statement with a value, in function returning 'void' [-fpermissive]
  import_array();
  ^
make[2]: *** [CMakeFiles/numpy_eigen.dir/src/autogen_module/numpy_eigen_export_module.cpp.o] Error 1
make[1]: *** [CMakeFiles/numpy_eigen.dir/all] Error 2
make: *** [all] Error 2
cd /home/naveen/kalibr_workspace/build/numpy_eigen; catkin build --get-env numpy_eigen | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << numpy_eigen:make                              [ Exited with code 2 ]
Failed    <<< numpy_eigen                                   [ 3.1 seconds ]    
Abandoned <<< aslam_imgproc                                 [ Unrelated job failed ]
Abandoned <<< sm_python                                     [ Unrelated job failed ]
Abandoned <<< aslam_cameras_april                           [ Unrelated job failed ]
Abandoned <<< aslam_cv_python                               [ Unrelated job failed ]
Abandoned <<< sparse_block_matrix                           [ Unrelated job failed ]
Abandoned <<< aslam_backend                                 [ Unrelated job failed ]
Abandoned <<< aslam_backend_expressions                     [ Unrelated job failed ]
Abandoned <<< aslam_backend_python                          [ Unrelated job failed ]
Abandoned <<< aslam_cv_backend                              [ Unrelated job failed ]
Abandoned <<< bsplines                                      [ Unrelated job failed ]
Abandoned <<< aslam_splines                                 [ Unrelated job failed ]
Abandoned <<< aslam_cv_error_terms                          [ Unrelated job failed ]
Abandoned <<< aslam_cv_backend_python                       [ Unrelated job failed ]
Abandoned <<< bsplines_python                               [ Unrelated job failed ]
Abandoned <<< aslam_splines_python                          [ Unrelated job failed ]
Abandoned <<< incremental_calibration                       [ Unrelated job failed ]
Abandoned <<< incremental_calibration_python                [ Unrelated job failed ]
Abandoned <<< kalibr                                        [ Unrelated job failed ]
```

#### CDE Package Errors
```
naveen@computer:~/Downloads/kalibr-cde$ ./kalibr_calibrate_cameras 
importing libraries
ImportError: numpy.core.multiarray failed to import
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/apport_python_hook.py", line 64, in apport_excepthook
    from apport.fileutils import likely_packaged, get_recent_crashes
ImportError: No module named apport.fileutils

Original exception was:
ImportError: numpy.core.multiarray failed to import
Traceback (most recent call last):
  File "/var/kalibr-build/devel/bin/kalibr_calibrate_cameras", line 5, in <module>
    exec(fh.read())
  File "<string>", line 6, in <module>
  File "/var/kalibr-build/devel/lib/python2.7/dist-packages/sm/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 1, in <module>
  File "/var/kalibr-build/devel/lib/python2.7/dist-packages/numpy_eigen/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 3, in <module>
ImportError: numpy.core.multiarray failed to import
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/apport_python_hook.py", line 64, in apport_excepthook
    from apport.fileutils import likely_packaged, get_recent_crashes
ImportError: No module named apport.fileutils

Original exception was:
Traceback (most recent call last):
  File "/var/kalibr-build/devel/bin/kalibr_calibrate_cameras", line 5, in <module>
    exec(fh.read())
  File "<string>", line 6, in <module>
  File "/var/kalibr-build/devel/lib/python2.7/dist-packages/sm/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 1, in <module>
  File "/var/kalibr-build/devel/lib/python2.7/dist-packages/numpy_eigen/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 3, in <module>
ImportError: numpy.core.multiarray failed to import
```

yzhou377, 2019-02-07:
Did you find any solution to this problem??

NaveenSehgal, 2019-02-07:
@yzhou377 Yeah, I had Anaconda installed on my computer and that messed up things with Kalibr's python packages and dependencies. I removed Anaconda and it worked fine

yzhou377, 2019-02-07:
Thank you for your replay Naveen, I did not have Anaconda installed on my ubuntu... have no idea why it is not working now...

closed 2019-04-27 by NaveenSehgal