ethz-asl/kalibr #234: Putting apriltags/aprilgrid on PyPI

safijari, 2018-10-16:
I use only the apriltags and aprilgrid stuff from this repo and roll my own calibration routines. It's a bit difficult to get Kalibr set up in new environments so I've been working to create binary python wheels for these in a single package that I will put on pypi. 

For now my repo just cloned the relevant code over, but in the future I Intend for kalibr to be a submodule. Do you guys have any interest in/opinions about this effort?

Note that my python wrapper is written from the ground up using pybind11 (and is not as comprehensive as the existing wrapper) to keep dependencies at a minimum and I'm building fully contained manylinux wheels meaning for most linux users getting these should be able to do a `pip install apriltags2_ethz`.

NikolausDemmel, 2018-11-06:
I'm not a maintainer, but I think this would be quite useful!

safijari, 2019-01-09:
It is on PyPI now https://pypi.org/project/apriltags2-ethz/

saulzar, 2021-01-22:
@safijari  This is great - do you also have a github repository for this somewhere?

safijari, 2021-01-22:
For posterity it's here: https://github.com/safijari/apriltags2_ethz

saulzar, 2021-02-03:
@safijari For what it's worth the reason it was slightly hard to discover as the PyPi page doesn't link back to your repo. Cheers!