ethz-asl/kalibr #235: Is this division by 4 necessary?

safijari, 2018-10-18:
https://github.com/ethz-asl/kalibr/blob/1223f9b8e189541498c251623e5f02bdb829d714/aslam_cv/aslam_cameras_april/src/GridCalibrationTargetAprilgrid.cpp#L131

As I understand `size()` just returns `rows*columns` which represents the number of tags and not the corners, and each `iter` is a single tag, so is that division by 4 throwing out perfectly fine tag detections?

NikolausDemmel, 2018-10-18:
If you look elsewhere it seems it is correct and `size()` refers to the points, not tags

https://github.com/ethz-asl/kalibr/blob/1223f9b8e189541498c251623e5f02bdb829d714/aslam_cv/aslam_cameras_april/src/GridCalibrationTargetAprilgrid.cpp#L79-L80

safijari, 2018-10-18:
Ah, I had missed the multiplication by 2 in the constructor here https://github.com/ethz-asl/kalibr/blob/1223f9b8e189541498c251623e5f02bdb829d714/aslam_cv/aslam_cameras_april/src/GridCalibrationTargetAprilgrid.cpp#L32

It all makes sense now. I did a pure python reimplementation and this point was causing me confusion as my implmenetation was working fine without the `/4`