ethz-asl/kalibr #246: Measures of a good calibration at the kalibr_camera_validator wikipage

MagnusOffermanns, 2018-12-01:
Hy everyone,

currently I try to validate my calibration with the _kalibr_camera_validator_ program. Since I am not using Kalibr regularly I am not sure how big the values are meant to be to be a good calibration.

I think it would be useful to add the borders in which the calibration is good to the _kalibr_camera_validator_ wiki page (https://github.com/ethz-asl/kalibr/wiki/calibration-validator). 

i.e:
mean_x < 7
std_x < 4
max_y < 3.14
min_x < 33 ...

IrisGreen, 2019-05-20:
Hello, would you mind to tell me how to do the validation?
I just open three terminals, in terminal 1 I start the ‘roscore’
Terminal 2:  kalibr_camera_validator —cam camchain.yaml —target april.yaml
Terminal 3: rosbag play awsome.bag -r 4.0 —topics /left /right
 But there was an error: Conversions is only valid for 1 or 2dimensions ,argument has 3 dimensions
I have tried a lot of times, but just stuck here.

ma7moudk, 2019-05-27:
> Hello, would you mind to tell me how to do the validation?
> I just open three terminals, in terminal 1 I start the ‘roscore’
> Terminal 2: kalibr_camera_validator —cam camchain.yaml —target april.yaml [...]

Most probably the problem is that the bag has color image topic . And this package needs a grayscale image "mono".

ma7moudk, 2019-05-27:
I have the same question . What is the criteria of a good calibration ? Which mean and standard deviation values are acceptable?