ethz-asl/kalibr #253: May be a bug in IccSensors.py

lishuaijielsj, 2019-01-07:
the findOrientationPriorCameraToImu compute a prior extrinsic form camera to imu, and initialize T_extrinsic.

but in initPoseSplineFromCamera  and addDesignVariables, T_extrinsic is used as transformation from imu to camera.

or could you explain the usage there if it is not a bug, Thanks!!!

icameling, 2019-01-29:
- 1. in `findOrientationPriorCameraToImu` , `T_extrinsic` is is initialized to Identity， so at this time, when `poseSpline = self.initPoseSplineFromCamera( timeOffsetPadding=0.0 )` is called, `posespline` is expressed in camera coordinate.
- 2. after `findOrientationPriorCameraToImu` is called, when you call `poseSpline = self.initPoseSplineFromCamera( timeOffsetPadding=0.0 )` again, at this time, `posespline` is **always** expressed in imu coordinate.

lishuaijielsj, 2019-02-11:
>     * 1. in `findOrientationPriorCameraToImu` , `T_extrinsic` is is initialized to Identity， so at this time, when `poseSpline = self.initPoseSplineFromCamera( timeOffsetPadding=0.0 )` is called, `posespline` is expressed in camera coordinate.
> 
>     * 1. after `findOrientationPriorCameraToImu` is called, when you call `poseSpline = self.initPoseSplineFromCamera( timeOffsetPadding=0.0 )` again, at this time, `posespline` is **always** expressed in imu coordinate.

lishuaijielsj, 2019-02-11:
> * 1. in `findOrientationPriorCameraToImu` , `T_extrinsic` is is initialized to Identity， so at this time, when `poseSpline = self.initPoseSplineFromCamera( timeOffsetPadding=0.0 )` is called, [...]

thanks, you are right. may be the name R_i_c in line 178
 R_i_c = q_i_c_Dv.toRotationMatrix().transpose() misleads me.
here, R_i_c means rotation from imu to camera.So, the T_extrinsic is from imu to camera.
Thanks again.

closed 2019-02-11 by lishuaijielsj