ethz-asl/kalibr #255: pose.initPoseSplineSparse(times, curve, knots, 1e-4) RuntimeError: [Exception]

WR0903, 2019-01-12:
wr@wr:~/kalibr_workspace/src/q$ kalibr_calibrate_imu_camera --target april_6x6_80x80cm.yaml --cam camchain.yaml --imu imu_adis16448.yaml --bag 2019-01-12-06-17-31.bag --bag-from-to 0 30
importing libraries
Initializing IMUs:
  Update rate: 70
  Accelerometer:
    Noise density: 0.01 
    Noise density (discrete): 0.0836660026534 
    Random walk: 0.0002
  Gyroscope:
    Noise density: 0.005
    Noise density (discrete): 0.0418330013267 
    Random walk: 4e-06
Initializing imu rosbag dataset reader:
	Dataset:          2019-01-12-06-17-31.bag
	Topic:            /imu/data
bagstart 1547273851.35
baglength 87.6256062984
[ WARN] [1547304655.818156]: BagImuDatasetReader: truncated 4109 / 6252 messages.
	Number of messages: 6252
Reading IMU data (/imu/data)
  Read 2143 imu readings over 30.0 seconds                   
Initializing calibration target:
  Type: aprilgrid
  Tags: 
    Rows: 6
    Cols: 6
    Size: 0.03 [m]
    Spacing 0.009 [m]
Initializing camera chain:
Camera chain - cam0:
  Camera model: pinhole
  Focal length: [314.6187799011879, 320.2956368270001]
  Principal point: [379.9776011718409, 225.8942986221469]
  Distortion model: equidistant
  Distortion coefficients: [0, 0, 0, 0]
  baseline: no data available
Initializing camera rosbag dataset reader:
	Dataset:          2019-01-12-06-17-31.bag
	Topic:            /frame
[ WARN] [1547304656.642365]: Bag end time of 30.0 s is bigger than the total length of 0.0 s
[ WARN] [1547304656.647354]: BagImageDatasetReader: truncated 0 / 2629 images.
	Number of images: 2629
Extracting calibration target corners
  Extracted corners for 2068 images (of 2629 images)                              

Building the problem
	Spline order: 6
	Pose knots per second: 100
	Do pose motion regularization: False
		xddot translation variance: 1000000.000000
		xddot rotation variance: 100000.000000
	Bias knots per second: 50
	Do bias motion regularization: True
	Blake-Zisserman on reprojection errors -1
	Acceleration Huber width (sigma): -1.000000
	Gyroscope Huber width (sigma): -1.000000
	Do time calibration: True
	Max iterations: 30
	Time offset padding: 0.030000
Estimating time shift camera to imu:

Initializing a pose spline with 0 knots (100.000000 knots per second over 0.000000 seconds)
Traceback (most recent call last):
  File "/home/wr/kalibr_workspace/devel/bin/kalibr_calibrate_imu_camera", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/wr/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera", line 236, in <module>
    main()
  File "/home/wr/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera", line 197, in main
    verbose = parsed.verbose)
  File "/home/wr/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccCalibrator.py", line 105, in buildProblem
    cam.findTimeshiftCameraImuPrior(self.ImuList[0], verbose)
  File "/home/wr/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccSensors.py", line 219, in findTimeshiftCameraImuPrior
    poseSpline = self.initPoseSplineFromCamera( timeOffsetPadding=0.0 )
  File "/home/wr/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccSensors.py", line 313, in initPoseSplineFromCamera
    pose.initPoseSplineSparse(times, curve, knots, 1e-4)
RuntimeError: [Exception] /home/wr/kalibr_workspace/src/Kalibr/aslam_nonparametric_estimation/bsplines/src/BSpline.cpp:970: initSplineSparse() assert(numSegments >= 1) failed [0 >= 1]: There must be at least one time segment

epiception, 2019-02-06:
Hey, any updates on this specific error? I noticed another issue open with the same problem.

mccoychang, 2020-08-27:
Hi, I also encountered the same problem as you. Can you share more details?

Pan-long-ming, 2021-07-07:
I also encountered the same problem as you,Can you show any details
Duplicate of #

goldbattle, 2022-04-22:
It looks like an invalid time range which means that there are no IMU measurements during this period.
```
[ WARN] [1547304656.642365]: Bag end time of 30.0 s is bigger than the total length of 0.0 s
```

closed 2022-05-04 by goldbattle