ethz-asl/kalibr #260: numpy_eigen installation problem on ubuntu 16.04

yzhou377, 2019-02-07:
I was trying to install Kalibr on my 16.04, but it failed at one package: numpy_eigen.

I am pretty sure that I have it properly installed but i still received the following warning:

`fatal error: numpy/arrayobject.h: No such file or directory
`
I have spent a few hours... please advise... Thank you!

jobesu14, 2019-02-12:
Same issue here, install fails on Ubuntu 16.04 when building numpy_eigen

jobesu14, 2019-02-12:
The solution is actually to read (and follow :-S) the installation process. I missed this point:

1. Install the build and run dependencies:

```sudo apt-get install python-setuptools python-rosinstall ipython libeigen3-dev libboost-all-dev doxygen libopencv-dev ros-indigo-vision-opencv ros-indigo-image-transport-plugins ros-indigo-cmake-modules python-software-properties software-properties-common libpoco-dev python-matplotlib python-scipy python-git python-pip ipython libtbb-dev libblas-dev liblapack-dev python-catkin-tools libv4l-dev```

```sudo pip install python-igraph --upgrade```

pLesur, 2019-04-01:
Issue persists even after installing some missing dependencies. I have ros-kinetic installed though, that might be why

pstcliminal, 2019-05-29:
I'm facing same problem. When I was tring to run kalibr_calibrate_imu on Python, three problems occured.
1. Python can’t find module sm
2. Python can’t find module numpy_eigen
3. Python can’t find module libnumpy_eigen
I found this site
https://pkok.github.io/2014/09/03/
so I solved two of them. But last one, libnumpy_eigen, has not been solved yet.
I grepped subfolders and found some files contain the word "libnumpy_eigen", but i can't tell how to solve it.

bloom256, 2019-06-06:
sudo apt-get install --reinstall python-numpy
It worked for me (Ubuntu 16.04, Kinetic)

wyw0424, 2020-04-15:
Try source devel/setup.bash and run it again. There is a /devel/lib/python2.7/dist-packages/numpy_eigen/ file in your ROS repo.

charchemist, 2022-01-19:
> I'm facing same problem on ROS kinetic on Ubuntu 16.04, Windows WSL. When I was tring to run kalibr_calibrate_imu with Python, three problems occured.
> 
> 1. Python can’t find module sm
> 2. Python can’t find module numpy_eigen
> 3. Python can’t find module libnumpy_eigen
>    I found this site
>    https://pkok.github.io/2014/09/03/
>    so I solved two of them. But last one, libnumpy_eigen, has not been solved yet.
>    I grepped subfolders and found some files contain the word "libnumpy_eigen", but I can't tell how to solve it.

excuse me, I got the third issue "Python can’t find module libnumpy_eigen" too, did you get the solution?thx

goldbattle, 2022-04-22:
Let me know if PR https://github.com/ethz-asl/kalibr/pull/515 still has this issue. Thanks!

closed 2022-04-22 by goldbattle