ethz-asl/kalibr #268: genmsg.msg_loader.MsgNotFound: Cannot locate message [Header]: unknown package [std_msgs] on search path [{}]

gakkiHp, 2019-03-02:
I calibrated the camera and ran the following command, but I got an error：
~/kalibr_workspace$ kalibr_calibrate_cameras     --target ap.yaml     --bag 8.bag     --models pinhole-fov     --topics /android/image_raw/compressed
importing libraries
Initializing cam0:
        Camera model:     pinhole-fov
        Dataset:          8.bag
        Topic:            /android/image_raw/compressed
Traceback (most recent call last):
  File "/home/nvidia/kalibr_workspace/devel/bin/kalibr_calibrate_cameras", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/nvidia/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_cameras", line 447, in <module>
    main()
  File "/home/nvidia/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_cameras", line 168, in main
    dataset = initBagDataset(parsed.bagfile, topic, parsed.bag_from_to)
  File "/home/nvidia/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_cameras", line 28, in initBagDataset
    reader = kc.BagImageDatasetReader(bagfile, topic, bag_from_to=from_to)
  File "/home/nvidia/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ImageDatasetReader.py", line 52, in __init__
    self.indices = self.sortByTime(self.indices)
  File "/home/nvidia/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ImageDatasetReader.py", line 63, in sortByTime
    topic, data, stamp = self.bag._read_message(self.index[idx].position)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbag/bag.py", line 986, in _read_message
    return self._reader.seek_and_read_message_data_record(position, raw)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbag/bag.py", line 2475, in seek_and_read_message_data_record
    msg_type = _get_message_type(connection_info)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbag/bag.py", line 1571, in _get_message_type
    message_type = genpy.dynamic.generate_dynamic(info.datatype, info.msg_def)[info.datatype]
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/dynamic.py", line 155, in generate_dynamic
    for l in msg_generator(msg_context, spec, search_path):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/generator.py", line 741, in msg_generator
    genmsg.msg_loader.load_depends(msg_context, spec, search_path)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genmsg/msg_loader.py", line 349, in load_depends
    return load_msg_depends(msg_context, spec, msg_search_path)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genmsg/msg_loader.py", line 318, in load_msg_depends
    depspec = load_msg_by_type(msg_context, resolved_type, search_path)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genmsg/msg_loader.py", line 125, in load_msg_by_type
    file_path = get_msg_file(package_name, base_type, search_path)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genmsg/msg_loader.py", line 84, in get_msg_file
    % (base_type, package, search_path), base_type, package, search_path)
genmsg.msg_loader.MsgNotFound: Cannot locate message [Header]: unknown package [std_msgs] on search path [{}]
what should i do?

diav79, 2019-03-26:
@gakkiHp  I am facing the same issue. Found any solution?

gakkiHp, 2019-03-27:
Maybe these links will help
https://github.com/gavanderhoorn/rosbag_fixer
https://github.com/ros/common_msgs/issues/20

gakkiHp, 2019-04-08:
@diav79  Have you solved it？

vishals1010, 2020-10-30:
Have you solved it？

yudhapane, 2020-12-11:
I have similar issue when trying to import a bag file and convert the data to a csv file. It works with the rosbag_fixer repository @SYSUHPK gave

xudooo, 2020-12-15:
I tried to calibrate Camera-Imu using an Android device, and got the same error: 
`genmsg.msg_loader.MsgNotFound: Cannot locate message [Header]: unknown package [std_msgs] on search path [{}]`.
Then I used [rosbag_fixer](https://github.com/gavanderhoorn/rosbag_fixer) motioned by @SYSUHPK and got another error:
`The time ranges of the camera and IMU do not overlap. Please make sure that your sensors are synchronized correctly.`
How can I sysnchronize the camera and imu? Thanks.

DandyLorenz, 2022-03-23:
Have you solved it？

DandyLorenz, 2022-03-26:
yes,Currently I have solved it. Reasons for errors include:
1.IMU information cannot be read directly by Kalib
2.Timestamps of IMU and CAMERA may not be in the same range
3.Timestamps of IMU and CAMERA may be out of sync

closed 2022-05-04 by goldbattle

david-delmoral, 2023-03-09:
> yes,Currently I have solved it. Reasons for errors include: 1.IMU information cannot be read directly by Kalib 2.Timestamps of IMU and CAMERA may not be in the same range 3.Timestamps of IMU and CAMERA may be out of sync

I am having the same issue and from your response I don't know how you go about solving each case. My IMU topic publishes at around 45 Hz, I don't know if such low frequency can be the cause.