ethz-asl/kalibr #269: Why is the re-projection error so large in  camera-IMU calibration?

GuaguaguaLiu, 2019-03-06:
Hello,
   
      I'm using the kalibr  to calibrate Camera-IMU.  But after many attempts, I still find that the re-projection error is very large(about 10). The device I used is MYNT-EYE (D1000)(stereo camera).   The attachment is the result of my cameras calibration and camera-IMU calibration(I have chosen the relatively good one ). Someone could  give me some advice? Thank you very much!

[report-imucam-1.pdf](https://github.com/ethz-asl/kalibr/files/2934198/report-imucam-1.pdf)
[report-imucam-1.pdf](https://github.com/ethz-asl/kalibr/files/2934201/report-imucam-1.pdf)
[image]
[image]

schneith, 2019-03-06:
The IMU data in the plots of the cam-IMU calibration look somewhat weird. Is it possible that there are very fast motions or shocks in the datasets? 

In the cam-cam calibration there are very few points close to the image corners (high distortion area) which would be quite informative for the calibration. If you use an Aprilgrid, you can record partial views of the target to ensure observations close to the image border.

Could you upload a video of your dataset?

GuaguaguaLiu, 2019-03-06:
Thank you very much for your reply! I calibrated it again, but the re-projection error is still large. Attached are the videos.
https://wx.qq.com/cgi-bin/mmwebwx-bin/webwxgetvideo?msgid=7939467719872439704&skey=%40crypt_6bf8eef_b765329ecd97222d0674b78c702f30f6

GuaguaguaLiu, 2019-03-06:
Thank you very much for your reply!   I calibrated it again, but the re-projection error was still large. 
I have attached the calibration videos(cam-cam and cam-imu). Look forward for your reply.
https://wx.qq.com/cgi-bin/mmwebwx-bin/webwxgetvideo?msgid=2317820790774638956&skey=%40crypt_6bf8eef_af22343ca7485e7b3b2877a8be958464

https://wx.qq.com/cgi-bin/mmwebwx-bin/webwxgetvideo?msgid=920173204439414472&skey=%40crypt_6bf8eef_af22343ca7485e7b3b2877a8be958464

GuaguaguaLiu, 2019-03-07:
Thank you very much for your reply! I calibrated it again, but the re-projection error was still large.
I have attached the calibration videos(cam-cam and cam-imu). Look forward for your reply.
[calibration_videos.tar.gz](https://github.com/ethz-asl/kalibr/files/2939440/calibration_videos.tar.gz)

shilei001, 2019-05-05:
Hi， how about the result，after Schneich‘s help？I used MYNTeye,the result is not good.Waiting for your reply.

EmiyaEstelle, 2019-09-29:
@shilei001 
Hi. I used MYNT EYE S too. And the re-projection error is from 2 to 4 and I just calibrated the binocular camera first. Did you solve the problem?