ethz-asl/kalibr #271: Odometry drifts off for stationary pose

Jwenda, 2019-03-25:
Dear community,
I'm facing a problem while running MSCKF in the Gazebo simulation. We have a Realsense D435 camera in the simulator running. I made a record of the running simulator and attached it to the post. The MSCKF is initially started from the static position on the ground, but the result is as same as started in the air.

I also attached the calibration file we used.
[out-2.zip](https://github.com/ethz-asl/kalibr/files/3002859/out-2.zip)
[realsense_simulation.txt](https://github.com/ethz-asl/kalibr/files/3002902/realsense_simulation.txt)

Do I miss something in the configuration file?

Thank you for your help!

Best,
Jan Bednar

Jwenda, 2019-03-25:
I'm sorry, I posted it to another source. I closed it.

closed 2019-03-25 by Jwenda

HappySamuel, 2019-04-22:
Hi @Jwenda 

I am using D435 in the gazebo simulation too. I am curious on how you do the calibration of d435 in the gazebo simulation? Please share with me.

Best.
Samuel

Jwenda, 2019-04-22:
Hi @HappySamuel 
actually, I haven|t calibrated the D435 in the Gazebo. I used the model that my colleagues made for the Gazebo. So I used the transformation the model gave to me without running the Kalibr. I wonder if anyone has tried to calibrate Gazebo camera. You have to create the Calibration targets in Gazebo, than it should be possible I think.

Best,
Jan