ethz-asl/kalibr #273: Android  camera and IMU calibration, Only camera is OK, but  Estimating imu-camera rotation prior

Jimsysu, 2019-04-03:
When I excute the  ./kalibr_calibrate_imu_camera , the error  as fllow:
importing libraries
Initializing IMUs:
  Update rate: 100.0
  Accelerometer:
    Noise density: 0.01 
    Noise density (discrete): 0.1 
    Random walk: 0.0002
  Gyroscope:
    Noise density: 0.005
    Noise density (discrete): 0.05 
    Random walk: 4e-06
Initializing imu rosbag dataset reader:
	Dataset:          /home/utopa/Project/kalibr-cde/Calibra_Android/meizuPhoneCameraIMU_april3_fix.bag
	Topic:            /android/imu
	Number of messages: 24663
Reading IMU data (/android/imu)
  Read 24663 imu readings over 262.6 seconds                         
Initializing calibration target:
  Type: aprilgrid
  Tags: 
    Rows: 6
    Cols: 6
    Size: 0.038 [m]
    Spacing 0.01254 [m]
Initializing camera chain:
Camera chain - cam0:
  Camera model: pinhole
  Focal length: [520.2206985217587, 519.4547165170969]
  Principal point: [320.4077221473467, 240.60631030735865]
  Distortion model: radtan
  Distortion coefficients: [0.08378577106966244, -0.1722614470117299, -0.0014135121035940901, 0.0001770202923349205]
  baseline: no data available
Initializing camera rosbag dataset reader:
	Dataset:          /home/utopa/Project/kalibr-cde/Calibra_Android/meizuPhoneCameraIMU_april3_fix.bag
	Topic:            /android/image_raw/compressed
	Number of images: 6212
Extracting calibration target corners
  Extracted corners for 2002 images (of 6212 images)                              

Building the problem
	Spline order: 6
	Pose knots per second: 70
	Do pose motion regularization: False
		xddot translation variance: 1000000.000000
		xddot rotation variance: 100000.000000
	Bias knots per second: 50
	Do bias motion regularization: True
	Blake-Zisserman on reprojection errors -1
	Acceleration Huber width (m/s^2): -1.000000
	Gyroscope Huber width (rad/s): -1.000000
	Do time calibration: False
	Max iterations: 30
	Time offset padding: 0.020000

Estimating imu-camera rotation prior

Initializing a pose spline with 26251 knots (100.000000 knots per second over 262.507000 seconds)
[FATAL] [1554285514.8730-target Calibra_Android/april_6x6_80x80cm.yaml  --bag Calibra_Android/meizuPhoneCameraIMU_april3_fix.bag  --cam Calibra_Android/MeiZ/camchain-homeutopaProjectkalibr-cdeCalibra_AndroidmeizuPhoneCameraImu_april_fix.yaml  --imu  Calibra_Android/MeiZ/imu_phone.yaml 
58]: Failed to obtain orientation  prior!
The target is :  april_6x6_80x80cm.yaml , is opened  on computer screen
Bag :  android apk written by ros, and advertise camera 15Hz and IMU 100 hz

well, I have calibration  a   camera and IMU sensor which the time Synchronizer is OK. 
But the phone is not OK, only  camera calibration is ok about this phone, but doesnot work with IMU, I think it is the time Synchronizer, I need to Synchronizer about the camera and imu ?

araujokth, 2019-04-25:
Hi @Jimsysu , any chance you could say which app you are using for recording the image and IMU data from the android device? Is it available online? I also wanted to give this a try.

Jimsysu, 2019-04-26:
hi, @araujokth https://github.com/hitcm/Android_Camera-IMU , it is used online,but  kalibr-cde calibration need the bagfile

gjgjh, 2019-09-29:
I have encountered the same problem, any solution yet?

gjgjh, 2019-09-29:
> I have encountered the same problem, any solution yet?

I think https://github.com/hitcm/Android_Camera-IMU has some bugs. Timestamps between IMU and image is not right, if you extract the raw data from the bag file. After synchronizing the data manually, I solve this problem finally.

mccoychang, 2020-07-07:
> 
> I think https://github.com/hitcm/Android_Camera-IMU has some bugs. Timestamps between IMU and image is not right, if you extract the raw data from the bag file. After synchronizing the data manually, I solve this problem finally.

Hi, how to manually sync data, please help. I have the same problem.

GabHoo, 2020-12-09:
https://github.com/OSUPCVLab/mobile-ar-sensor-logger

Hi, I found this one which works amazing for me. Best app.

Still need to find a good camera-IMU transformation matrix tho. I am using a Huawei P20 smartphone.

goldbattle, 2022-04-27:
Both need to be in the same clock, but can have a temporal timeoffset / delay. If you can convert the data from the phone into a rosbag then Kalibr should work as expected.