ethz-asl/kalibr #276: 1 packages failed and 12 packages were abandoned with latest master code

mysunnyday, 2019-04-26:
grw@ubuntu:~/kalibr_workspace$ catkin build -DCMAKE_BUILD_TYPE=Release -j4
---------------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/ros/indigo
Workspace:                   /home/grw/kalibr_workspace
---------------------------------------------------------------
Build Space:        [exists] /home/grw/kalibr_workspace/build
Devel Space:        [exists] /home/grw/kalibr_workspace/devel
Install Space:      [unused] /home/grw/kalibr_workspace/install
Log Space:          [exists] /home/grw/kalibr_workspace/logs
Source Space:       [exists] /home/grw/kalibr_workspace/src
DESTDIR:            [unused] None
---------------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
---------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        -j4
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
---------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
---------------------------------------------------------------
Workspace configuration appears valid.
---------------------------------------------------------------
[build] Found '37' packages in 0.0 seconds.                                     
[build] Package table is up to date.                                            
Starting  >>> catkin_simple                                                     
Starting  >>> ethz_apriltag2                                                    
Starting  >>> python_module                                                     
Starting  >>> sm_common                                                         
Finished  <<< python_module                                 [ 0.4 seconds ]     
Starting  >>> sm_logging                                                        
Finished  <<< ethz_apriltag2                                [ 0.6 seconds ]     
Starting  >>> sm_random                                                         
Finished  <<< sm_common                                     [ 0.4 seconds ]     
Starting  >>> suitesparse                                                       
Finished  <<< catkin_simple                                 [ 0.4 seconds ]     
Starting  >>> numpy_eigen                                                       
Finished  <<< sm_logging                                    [ 0.4 seconds ]     
Starting  >>> sm_boost                                                          
Finished  <<< sm_random                                     [ 0.4 seconds ]     
Starting  >>> sm_matrix_archive                                                 
Finished  <<< suitesparse                                   [ 0.5 seconds ]     
Starting  >>> sm_opencv                                                         
Finished  <<< sm_boost                                      [ 0.5 seconds ]     
Finished  <<< sm_matrix_archive                             [ 0.5 seconds ]     
Starting  >>> sm_property_tree                                                  
Finished  <<< sm_opencv                                     [ 0.4 seconds ]     
Starting  >>> aslam_time                                                        
Finished  <<< sm_property_tree                              [ 0.4 seconds ]     
Starting  >>> opencv2_catkin                                                    
Finished  <<< aslam_time                                    [ 0.5 seconds ]     
Starting  >>> sm_eigen                                                          
Finished  <<< opencv2_catkin                                [ 0.4 seconds ]     
Starting  >>> sm_timing                                                         
Finished  <<< numpy_eigen                                   [ 2.2 seconds ]     
Finished  <<< sm_eigen                                      [ 0.5 seconds ]     
Starting  >>> sm_kinematics                                                     
Starting  >>> sparse_block_matrix                                               
Finished  <<< sm_timing                                     [ 0.4 seconds ]     
Finished  <<< sm_kinematics                                 [ 0.5 seconds ]     
Starting  >>> aslam_cameras                                                     
Starting  >>> sm_python                                                         
Finished  <<< sm_python                                     [ 0.5 seconds ]     
Finished  <<< aslam_cameras                                 [ 0.6 seconds ]     
Starting  >>> aslam_cv_serialization                                            
Starting  >>> aslam_imgproc                                                     
[... 49 lines cut ...]
                 from /home/grw/kalibr_workspace/src/Kalibr/aslam_optimizer/aslam_backend/include/aslam/backend/Marginalizer.hpp:11,
                 from /home/grw/kalibr_workspace/src/Kalibr/aslam_optimizer/aslam_backend/src/Marginalizer.cpp:8:
/home/grw/kalibr_workspace/src/Kalibr/aslam_optimizer/aslam_backend/src/Marginalizer.cpp: In function ‘void aslam::backend::marginalize(std::vector<aslam::backend::DesignVariable*>&, std::vector<aslam::backend::ErrorTerm*>&, int, bool, boost::shared_ptr<aslam::backend::MarginalizationPriorErrorTerm>&, Eigen::MatrixXd&, std::vector<aslam::backend::DesignVariable*>&, std::size_t, std::size_t)’:
/home/grw/kalibr_workspace/src/Kalibr/Schweizer-Messer/sm_common/include/sm/assert_macros.hpp:121:27: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   if((value) < (lowerBound))            \
                           ^
/home/grw/kalibr_workspace/src/Kalibr/aslam_optimizer/aslam_backend/src/Marginalizer.cpp:166:6: note: in expansion of macro ‘SM_ASSERT_GE’
      SM_ASSERT_GE(aslam::Exception, R.rows(), numTopRowsInCov, "Cannot extract " << numTopRowsInCov << " rows of R because it only has " << R.rows() << " rows.");
      ^
/home/grw/kalibr_workspace/src/Kalibr/Schweizer-Messer/sm_common/include/sm/assert_macros.hpp:121:27: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   if((value) < (lowerBound))            \
                           ^
/home/grw/kalibr_workspace/src/Kalibr/aslam_optimizer/aslam_backend/src/Marginalizer.cpp:167:6: note: in expansion of macro ‘SM_ASSERT_GE’
      SM_ASSERT_GE(aslam::Exception, R.cols(), numTopRowsInCov, "Cannot extract " << numTopRowsInCov << " cols of R because it only has " << R.cols() << " cols.");
      ^
cd /home/grw/kalibr_workspace/build/aslam_backend; catkin build --get-env aslam_backend | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
................................................................................
Finished  <<< aslam_backend                                 [ 3 minutes and 49.9 seconds ]
[build] Summary: 24 of 37 packages succeeded.                                   
[build]   Ignored:   None.                                                      
[build]   Warnings:  1 packages succeeded with warnings.                        
[build]   Abandoned: 12 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                         
[build] Runtime: 4 minutes and 15.4 seconds total.                              
[build] Note: Workspace packages have changed, please re-source setup files to use them.

mysunnyday, 2019-04-26:
I found solution , input command below then build again . I think the reason is I have interrupt the build once
> cd /home/grw/kalibr_workspace/build/aslam_cv_python; catkin build --get-env aslam_cv_python | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -

closed 2019-04-26 by mysunnyday

nb-wang, 2021-06-27:
excuse me, i still meet the error. is there any idea to solve this? thx!

SizheWei, 2022-03-04:
> I found solution , input command below then build again . I think the reason is I have interrupt the build once
> 
> > cd /home/grw/kalibr_workspace/build/aslam_cv_python; catkin build --get-env aslam_cv_python | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -

I also met this problem, and your solution works for me. Thank you!

egemenertugrul, 2022-12-12:
Those who get "c++: fatal error: Killed signal terminated program cc1plus" can apply the solution here: https://github.com/soedinglab/hh-suite/issues/280#issuecomment-1076146671
It worked for me.