ethz-asl/kalibr #277: bagextractor cannot find the bag file [cde]

ahmadhasan2k8, 2019-04-26:
To test the cde-tools I downloaded MH_01_easy dataset. Now I can create a bag from the dataset using 
```
./kalibr_bagcreater --folder /home/ahasan/Downloads/MH_01_easy/mav0/. --output-bag awesome.bag
```
But then, if I try to extract the bag file with 
```
./kalibr_bagextractor --image-topics /cam0/image_raw /cam1/image_raw --imu-topics /imu0  --bag awesome.bag 
```
Surprisingly my bag file gets deleted and I get the following error

```
importing libraries
the rosdep view is empty: call 'sudo rosdep init' and 'rosdep update'

Traceback (most recent call last):
  File "/var/kalibr-build/devel/bin/kalibr_bagextractor", line 5, in <module>
    exec(fh.read())
  File "<string>", line 46, in <module>
  File "/var/kalibr-build/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ImageDatasetReader.py", line 29, in __init__
    self.bag = rosbag.Bag(bagfile)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/rosbag/bag.py", line 159, in __init__
    self._open(f, mode, allow_unindexed)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/rosbag/bag.py", line 890, in _open
    if   mode == 'r': self._open_read(f, allow_unindexed)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/rosbag/bag.py", line 903, in _open_read
    self._file     = open(f, 'rb')
IOError: [Errno 2] No such file or directory: 'awesome.bag'
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/apport_python_hook.py", line 64, in apport_excepthook
    from apport.fileutils import likely_packaged, get_recent_crashes
ImportError: No module named apport.fileutils

Original exception was:
Traceback (most recent call last):
  File "/var/kalibr-build/devel/bin/kalibr_bagextractor", line 5, in <module>
    exec(fh.read())
  File "<string>", line 46, in <module>
  File "/var/kalibr-build/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ImageDatasetReader.py", line 29, in __init__
    self.bag = rosbag.Bag(bagfile)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/rosbag/bag.py", line 159, in __init__
    self._open(f, mode, allow_unindexed)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/rosbag/bag.py", line 890, in _open
    if   mode == 'r': self._open_read(f, allow_unindexed)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/rosbag/bag.py", line 903, in _open_read
    self._file     = open(f, 'rb')
IOError: [Errno 2] No such file or directory: 'awesome.bag'
```

ahmadhasan2k8, 2019-04-29:
The problem comes from here 
```
#delete old results
rm -f $HERE/*.pdf 2> /dev/null
rm -f $HERE/*.yaml 2> /dev/null
rm -f $HERE/*.txt 2> /dev/null
rm -f $HERE/*.bag 2> /dev/null
rm -rf $HERE/cde-root/output* 2> /dev/null

```

So, if I put the awesome.bag in a different folder. It works.

goldbattle, 2022-05-04:
CDE is no longer supported.

closed 2022-05-04 by goldbattle