ethz-asl/kalibr #278: imu-camera calib for rolling shutter camera is still not working,right?

mysunnyday, 2019-04-28:
hi , thank your great work in this project.
I know we can calibrate the rolling shutter camera with kalibr_calibrate_rs_cameras, that's great...
but what can I do for the imu-camera calibration with this result. 
I'm trying to calibr a android phone, I didn't found any parameter in kalibr_calibrate_imu_camera which could use the result from kalibr_calibrate_rs_camera.
I want to calib the misalignment-scale parameter of imu . I think the rolling effect of feature could spoil this extend method,right ?
so any advice ? :)

mysunnyday, 2019-04-29:
it seems some error about time stamp, I write the image to sdcard in the ndk camera2 callback function. but I found the timestamp is not stable, most stamp gap between sequence frame is 33ms , but sometime it could be 66ms or 99ms, I don't know if this situation would broken the optimizer ?

mysunnyday, 2019-04-29:
and in the ndk code , I have set the ACAMERA_CONTROL_AE_TARGET_FPS_RANGE to 30 then get 30fps , 
and in the callback func I can get ACAMERA_SENSOR_ROLLING_SHUTTER_SKEW as 33ms , and ACAMERA_SENSOR_EXPOSURE_TIME around 9ms ~ 29ms in different light condition.  so what's the relationship between exposure time, rolling sutter time, and frame rate and line delay? have I used the right camera configuration ?

mysunnyday, 2019-04-29:
anyway , use the non-rs method like 
kalibr_calibrate_cameras --target april_6x6.yaml --bag cam0s2_sample.bag --models pinhole-radtan --topics /cam0/image_raw

always return a result, although the re-proj error is bounded in 1pix, I think it's not good enough, I'd like to try the rolling edition for improvement

goldbattle, 2019-10-17:
For others that run into this issue, I would recommend you checkout the pr https://github.com/ethz-asl/kalibr/pull/261 if you have not already.

SjjAugust, 2020-03-19:
I have the same problem. Did you figure out it ?

goldbattle, 2020-03-19:
@SjjAugust have you tried pr #261?

SjjAugust, 2020-03-19:
> @SjjAugust have you tried pr #261?

Yes, I follow pr # 261, here is my operation
1. Keep the camera still, move the target quickly, record a 40-second video (30hz)
2. Get each frame of the video and timestamp it
3. Create a bag using bagcreater
4. Use the following script to calibrate `python kalibr_calibrate_rs_cameras \
--model pinhole-equi-rs --target april_6x6_80x80cm.yaml \
--topic / cam0 / image_raw \
--inverse-feature-variance 1 --frame-rate 30 `
After extracting the corner points of all pictures, a problem occurred: `list index out of range`
I don't know what went wrong, hope to get help, thanks.

goldbattle, 2020-03-19:
Did you checkout the PR branch? or just the master in this repository?
```
git fetch origin pull/261/head:pr_261
git checkout pr_261
git log -1 <- should have commit from that pr
```
You should be able to run it with rosrun after rebuilding in your workspace.

SjjAugust, 2020-03-20:
> Did you checkout the PR branch? or just the master in this repository?
> 
> ```
> git clone https://github.com/ethz-asl/kalibr/
> cd kalibr
> git fetch origin pull/261/head:pr_261
> git checkout pr_261
> git log -1 <- should have commit from that pr
> ```
> 
> You should be able to run it with rosrun after rebuilding in your workspace.

Thank you very much, I can run smoothly after changing the branch, here are my results:
LineDelay:
3.08641975309e-05
Intrinsics:
[ 2553.76736216  2553.76736216   959.5          539.5       ]
Distortion:
[ 0.  0.  0.  0.]
Using the block_cholesky linear system solver
Using the dog_leg trust region policy
Using the block_cholesky linear system solver
Using the dog_leg trust region policy
Initializing
Optimization problem initialized with 428480 design variables and 9039 error terms
The Jacobian matrix is 18077 x 2570871
[0.0]: J: 86229.5

I still have two questions:
1. I use 30hz to shoot 1080p video, but the obtained linedelay is exactly equal to 1 / (30 * 1080). Does this prove that the selected picture has not been skewed during calculation? I check every frame of the video. I found that the skewed pictures are blurry. Is it related to this?
2. After getting the above results, the console is stuck. Does this indicate that the calculation has been completed? Is this normal?

goldbattle, 2020-03-20:
What is printed should be the initial guess. It should optimize for quite some time, and that first `J:` printout is just before it starts doing the first iteration if my memory is correct. It also has a knot refinement stage also which means it takes 2x as long to do just the first problem.

goldbattle, 2022-06-08:
Feel free to reopen with a bag to reproduce any problems.

closed 2022-06-08 by goldbattle