ethz-asl/kalibr #284: CDE package don't work

pstcliminal, 2019-05-29:
I am having trouble using Kalibr CDE package on Ubuntu 16.04. 

Following are command history. Same as Installation page.
`tar xfvz kalibr_20062014.tar.gz`
`export PATH="/home/xxx/kalibr-cde:$PATH"`

result:
`cde-exec: cde.c:2595: finish_setup_shmat: Assertion 'cur_regs.orig_rax == 30' failed.`
`strace: exec: Function not implemented`
`./kalibr_calibrate_imu_camera: line 59:   183 Aborted                 (core dumped) ../cde-exec '-o' $CDE_PATH $CDE_TARGET_SCRIPT $arg_string`

Anyone have some ideas?

jaymayn, 2019-09-10:
I'm having a similar issue with executing cde-exec, were you able to resolve it.. or do you have any further insights

goldbattle, 2019-10-17:
The CDE hasn't been updated since 2014, I would recommend building from master instead of using it.