ethz-asl/kalibr #285: Failed << aslam_cv_serialization

chenyumingx, 2019-05-30:
Hi all,
I followed the tutorial exactly, when I execute the following command
> catkin build -DCMAKE_BUILD_TYPE=Release -j4

**error occurred！**
> /home/hnu/kalibr_ws/src/Kalibr/aslam_cv/aslam_cv_serialization/src/autogen/Camera-PinholeCameraGeometry.cpp:14:68:   required from here
/usr/include/boost/archive/detail/iserializer.hpp:237:47: error: invalid conversion from ‘long unsigned int’ to ‘void*’ [-fpermissive]
                 (T::operator delete)(t, sizeof(T));

The terminal keeps displaying the following information：
> [build -27:06.8] [21/37 complete] [1/4 jobs] [0 queued] [1 failed] [suitesparse:make (20%) - 28:18.3]

I also refered to this issue #236

closed 2019-06-19 by chenyumingx

Maxinjun, 2019-07-24:
@chenyumingx 
请问下这个问题你怎么解决的?我也遇到相同的问题

Wanghy118, 2022-05-13:
hi, i met the same problem, how do you solve this problem?