ethz-asl/kalibr #288: Did not coverge in maxIterations - Multicam Calibration

arjunds, 2019-06-20:
[image]
[image]

So for some reason when I try to calibrate my depth and stereo cameras together, it fails, but when I do each of them separately, it works perfectly fine. Attached are screenshots for what happens when I try to calibrate it together. I have tried increase the maxIterations value from 50 all the way up to 500, but still with no luck.

Anyone have this issue?

Current-River, 2019-08-04:
excuse me,I have the same problem.have you solved it?

arjunds, 2019-08-05:
The only solution I found was to use IMU-Camera calibration and that yielded better results.

closed 2019-08-05 by arjunds

liumingzs, 2023-12-21:
I move the d435i camera slowly and avoid turn the camera big degree