ethz-asl/kalibr #290: Error while kalibr_calibrate_rs_cameras

vhanded, 2019-06-26:
Hi,

Thanks for the rolling shutter calibration implementation. I run this command:

`rosrun kalibr kalibr_calibrate_rs_cameras --model pinhole-radtan-rs --target apriltag.yaml --topic /camera/infra1/image_rect_raw --bag rs_cam_hz4.bag --inverse-feature-variance 1 --frame-rate 30 --verbose --show-extraction` 

And at the end, I got this result:

```
Extracted corners for 1460 images (of 1460 images)                              

Initializing a pose spline with 506 knots (100.000000 knots per second over 50.612438 seconds)
[ERROR] [1561514699.257967]: Exception: list index out of range
```
I had run it several times with different recorded bag, but same result.

I can't find much solution in the github issue, any help?

goldbattle, 2019-10-17:
For others that run into this issue, I would recommend you checkout the pr https://github.com/ethz-asl/kalibr/pull/261 if you have not already.

goldbattle, 2022-04-22:
Should be fixed via #261 and #416

closed 2022-04-22 by goldbattle