ethz-asl/kalibr #296: ds-none model not found using CDE-pkg

EnriqueSolarte, 2019-08-23:
Hi everybody, 
I want to confirm something. Is 'ds-none' model supported for Kalibr-CDE 64-bit. 
I got the following error:
`
 File "/var/kalibr-build/devel/bin/kalibr_calibrate_cameras", line 5, in <module>
    exec(fh.read())
  File "<string>", line 444, in <module>
  File "<string>", line 190, in main
RuntimeError: Unknown camera model: ds-none. Try ['pinhole-radtan', 'pinhole-equi', 'omni-radtan'].
`

In advance thank you so much!

NikolausDemmel, 2019-08-23:
No, CDE doesn't support it. The [download page](https://github.com/ethz-asl/kalibr/wiki/downloads) says it was last updated in 2014. `ds-none` was added about a year ago. You'll have to compile Kalibr `master` branch.

You can also try [basalt's calibration tool](https://gitlab.vision.in.tum.de/basalt/basalt/blob/master/doc/Calibration.md#camera-calibration), which also has a [binary installer](https://gitlab.vision.in.tum.de/basalt/basalt#apt-installation-for-ubuntu-1604-and-1804-fast) for Ubuntu 16.04 and 18.04.

goldbattle, 2022-04-22:
Don't use the CDE, build from master, thanks!.