ethz-asl/kalibr #298: How to choose the models when using kalibr_calibrate_cameras

DingZiming, 2019-09-14:
I want to use kalibr_calibrate_cameras to calibrate my stereo camera (D:112° H:95° V:50°). And I moved my camera to captured the _Aprilgrid target_ for the bag file (dynamic dataset). 
 I used
 `kalibr_calibrate_cameras --bag camera.bag --topic /cam0/image_raw /cam1/image_raw --models omni-radtan omni-radtan  --target april_6x6.yaml `
but I got an error: _" invalid xi_omni of -0.137365150587 (xi>0)"_, and I changed another model _pinhole-radtan_ and it also failed. Finally, I use _pinhole-equi_ model and it worked. So I wanted to know about how to choose the models?
Thanks!

DingZiming, 2019-09-15:
I made a another bag (static dataset) and I succeed in calibrating the stereo camera with the _omni-radtan_ model. And anyone can give me some suggestions for the model choosing?

NikolausDemmel, 2019-09-20:
You need to make sure you get low reprojection error (< 0.5, ideally less), and cover the whole image. For both you can check the pdf report. It might be that you get low error, but you are only covering the center of the image, then that is not useful either. Check the tutorial for calibration in the wiki.

From my experience: pinhole-radtan is a good general purpose model, if you don't have too much distortion; it models a bit of radial and tangential distortion. pinhole-equi is great for many lenses, including some fisheye, if you only have radial distortion (it doesn't model tangential). Omni-radtan fits many fisheye lenses okeyish. ds and eucm are great and compact (few parameters) for some fisheye lenses, but also don't model any tangential distortion. pinhole fov is compact, and works ok for some medium fov fisheye lenses. In general Kalibr sometimes has trouble with initializing the optimization for too-wide FOV images, so it cannot calibrate, even though the model might still fit the lens ok.

DingZiming, 2019-10-18:
Thanks a lot !

closed 2019-10-18 by DingZiming