ethz-asl/kalibr #299: 1 Failed(suitesparse) and 28 abandoned in catkin build step

DoubleWzx, 2019-09-16:
**Details:**
Errors     << suitesparse:make /home/ring/kalibr_workspace/logs/suitesparse/build.make.000.log
--2019-09-16 06:46:55--  http://faculty.cse.tamu.edu/davis/SuiteSparse/SuiteSparse-4.2.1.tar.gz
正在解析主机 faculty.cse.tamu.edu (faculty.cse.tamu.edu)... 128.194.138.45
正在连接 faculty.cse.tamu.edu (faculty.cse.tamu.edu)|128.194.138.45|:80... 已连接。
已发出 HTTP 请求，正在等待回应... 302 Found
位置：http://211.81.63.2/cache/1/02/faculty.cse.tamu.edu/b94bd0cfbf97933f12a12efe98deaa79/SuiteSparse-4.2.1.tar.gz [跟随至新的 URL]
--2019-09-16 06:46:55--  http://211.81.63.2/cache/1/02/faculty.cse.tamu.edu/b94bd0cfbf97933f12a12efe98deaa79/SuiteSparse-4.2.1.tar.gz
正在连接 211.81.63.2:80... 已连接。
已发出 HTTP 请求，正在等待回应... 200 OK
长度： 22541409 (21M) [application/octet-stream]
正在保存至: “SuiteSparse-4.2.1.tar.gz”

SuiteSparse-4.2.1.t 100%[===================>]  21.50M   843KB/s    in 27s     

2019-09-16 06:47:22 (818 KB/s) - 已保存 “SuiteSparse-4.2.1.tar.gz” [22541409/22541409])
tar: 跳转到下一个头

gzip: stdin: invalid compressed data--crc error

gzip: stdin: invalid compressed data--length error
tar: Child returned status 1
tar: Error is not recoverable: exiting now
make[2]: *** [suitesparse_src-prefix/src/suitesparse_src-stamp/suitesparse_src-patch] Error 2
make[1]: *** [CMakeFiles/suitesparse_src.dir/all] Error 2
make: *** [all] Error 2
cd /home/ring/kalibr_workspace/build/suitesparse; catkin build --get-env suitesparse | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
Failed     << suitesparse:make                              [ Exited with code 2 ]
Failed    <<< suitesparse                                   [ 30.2 seconds ]   
Abandoned <<< sm_boost                                      [ Unrelated job failed ]
Abandoned <<< sm_matrix_archive                             [ Unrelated job failed ]
Abandoned <<< sm_opencv                                     [ Unrelated job failed ]
Abandoned <<< sm_property_tree                              [ Unrelated job failed ]
Abandoned <<< sm_eigen                                      [ Unrelated job failed ]
Abandoned <<< sm_timing                                     [ Unrelated job failed ]
Abandoned <<< sm_kinematics                                 [ Unrelated job failed ]
Abandoned <<< aslam_cameras                                 [ Unrelated job failed ]
Abandoned <<< aslam_cv_serialization                        [ Unrelated job failed ]
Abandoned <<< aslam_imgproc                                 [ Unrelated job failed ]
Abandoned <<< sm_python                                     [ Unrelated job failed ]
Abandoned <<< aslam_cameras_april                           [ Unrelated job failed ]
Abandoned <<< aslam_cv_python                               [ Unrelated job failed ]
Abandoned <<< sparse_block_matrix                           [ Unrelated job failed ]
Abandoned <<< aslam_backend                                 [ Unrelated job failed ]
Abandoned <<< aslam_backend_expressions                     [ Unrelated job failed ]
Abandoned <<< aslam_backend_python                          [ Unrelated job failed ]
Abandoned <<< aslam_cv_backend                              [ Unrelated job failed ]
Abandoned <<< bsplines                                      [ Unrelated job failed ]
Abandoned <<< aslam_splines                                 [ Unrelated job failed ]
Abandoned <<< aslam_cv_error_terms                          [ Unrelated job failed ]
Abandoned <<< aslam_cv_backend_python                       [ Unrelated job failed ]
Abandoned <<< bsplines_python                               [ Unrelated job failed ]
Abandoned <<< aslam_splines_python                          [ Unrelated job failed ]
Abandoned <<< incremental_calibration                       [ Unrelated job failed ]
Abandoned <<< incremental_calibration_python                [ Unrelated job failed ]
Abandoned <<< kalibr                                        [ Unrelated job failed ]

**how to solve it?
help me 
thanks**

MohrMa2, 2019-09-17:
> 2019-09-16 06:47:22 (818 KB/s) - 已保存 “SuiteSparse-4.2.1.tar.gz” [22541409/22541409])
> tar: 跳转到下一个头
> 
> gzip: stdin: invalid compressed data--crc error
> 
> gzip: stdin: invalid compressed data--length error
> tar: Child returned status 1
> tar: Error is not recoverable: exiting now

The SuiteSparse Archive is corrupted. As this is needed, nothing gets compiled.

The original file (http://faculty.cse.tamu.edu/davis/SuiteSparse/SuiteSparse-4.2.1.tar.gz) extracts here without errors, but yours gets cached (http://211.81.63.2/cache/1/02/faculty.cse.tamu.edu/b94bd0cfbf97933f12a12efe98deaa79/SuiteSparse-4.2.1.tar.gz) maybe by an proxy?

DoubleWzx, 2019-09-23:
> The SuiteSparse Archive is corrupted. As this is needed, nothing gets compiled.
>
> The original file (http://faculty.cse.tamu.edu/davis/SuiteSparse/SuiteSparse-4.2.1.tar.gz) extracts here without [...]

I don't konw about proxy. Do you have any idea about soving it?

EmiyaEstelle, 2019-10-09:
Hi, I meet the same problem like you. 
I guess that the source code of Kalibr needs to download the package "suitespares". You can see the CMakeLists in the "suitesparse" folder 
`DOWNLOAD_COMMAND rm -f SuiteSparse-${VERSION}.tar.gz && wget http://faculty.cse.tamu.edu/davis/SuiteSparse/SuiteSparse-${VERSION}.tar.gz`
The first time I tried to catkin build the Kalibr was wrong. It took me a long time to download the suitesparse. After it was done, I tried it again and it could work.
Hope to help you.

Nronaldo, 2020-05-13:
> Hi, I meet the same problem like you.
> I guess that the source code of Kalibr needs to download the package "suitespares". You can see the CMakeLists in the "suitesparse" folder
> `DOWNLOAD_COMMAND rm [...]

hello, could you please tell me how to download the suitesparse?

EmiyaEstelle, 2020-05-13:
@Nronaldo 
Hello.
I can hardly remember how to solve it.  Just what I said you can see above. When you install the kalibr with the way of WIKI, it complies in some parts and it took a long time to download suitesparse. I mean the download is during the installation.

Nronaldo, 2020-05-15:
> @Nronaldo
> Hello.
> I can hardly remember how to solve it. Just what I said you can see above. When you install the kalibr with the way of WIKI, it complies in some parts and it took a long time to download suitesparse. I mean the download is during the installation.

Hello. Thanks for your reply. I compile it and fail, so I watch this issue for help.  Then I compile it again and succeed.

helenol, 2020-07-20:
If anyone is hitting this now (looks like the webserver is currently down), you can replace the path in suitesparse/CMakeLists.txt with https://mirror.sobukus.de/files/src/suitesparse/ .

closed 2022-05-03 by goldbattle