ethz-asl/kalibr #303: What does the "image index" mean?

EmiyaEstelle, 2019-09-29:
Hi
Recently I have been using the Kalibr to do the imu-cam calibration. I have got a result (you can see it in the below picture)
[image]
So what meanings of "image index" is it? Waht meanings does the value of this index have?
Thank you

goldbattle, 2019-10-17:
My understanding is that image index should correspond to the images collected through the dataset.
So zero would be the start of the dataset, with larger ones being towards the end.
Note that there is frame selection, so it is not a 1-to-1 with the images in the rosbag normally.

EmiyaEstelle, 2019-10-18:
Thanks for your answer @goldbattle 
You mean that it is aussmed that I collect 100 pictures from a camera. But actually no every picture is good for kalibr or that is to say that kalibr will choose the appropriate picture in a rosbag. So the image index is the one which are choosen.

closed 2022-04-27 by goldbattle