ethz-asl/kalibr #305: No module named "libnumpy_eigen"

XiongLianga, 2019-10-10:
When I try to calibrate the camera, I run the calibr_calibrate_cameras. However, I get the error:
importing libraries
Failed to load Python extension for LZ4 support. LZ4 compression will not be available.
Traceback (most recent call last):
  File "/home/john/kalibr_workspace/devel/bin/kalibr_calibrate_cameras", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/john/kalibr_workspace/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_cameras", line 6, in <module>
    import sm
  File "/home/john/kalibr_workspace/devel/lib/python2.7/dist-packages/sm/__init__.py", line 35, in <module>
    exec(__fh.read())
  File "<string>", line 1, in <module>
  File "/home/john/kalibr_workspace/devel/lib/python2.7/dist-packages/numpy_eigen/__init__.py", line 35, in <module>
    exec(__fh.read())
  File "<string>", line 3, in <module>
ModuleNotFoundError: No module named 'libnumpy_eigen'

XiongLianga, 2019-10-10:
It consumes my some hours, lookfoward your solution.

oafolabi, 2019-11-25:
@XiongLianga  , I ran into the same issue. I reverted to using python2. I think the problem with my set up was that I had anaconda installed and so its python kept getting called as the default python. When I switched to python two on my system without anaconda, it was fine.

To switch you can change the line "#!/usr/bin/env python" to "#!/usr/bin/env python2" at the beginning of kalibr_workspace/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_cameras.py . Hope that helps

charchemist, 2022-01-19:
Excuse me, did you get the solution to this problem? I changed the line "#!/usr/bin/env python" to "#!/usr/bin/env python2" , but it doesn't work. looking forward to other solution, thx

goldbattle, 2022-04-22:
Please try PR https://github.com/ethz-asl/kalibr/pull/515 to see if that fixes your issues.

closed 2022-05-03 by goldbattle

shixiaosongye, 2024-10-29:
hello，I meet this question too,but  my python3and python2 all have  ModuleNotFoundError 'libnumpy_eigen',could you help me?