ethz-asl/kalibr #306: Errors of basline in the camera calibration

EmiyaEstelle, 2019-10-10:
Hi, I am using kalibr to calibrate a binocular camera with a IMU in it.
I use the april grid to do the calibation and everything is ok except the baseline. I get the baseline from the camera company that it is 120mm. But I try a few times to use kalibr to only do the camera calibration and their results are all almost 140mm.
The both reprojection errors are below 1 pixel. Perheps it is not very good. I choose the best one of results to copy there and hope some one could give some suggestions,
Thank you
[image]
[image]

floriantschopp, 2019-10-10:
What is the resolution of your cameras? Normally it is a good practice to try to cover the whole image with detections of the calibration board. In your case I see that only a small fraction of the image is covered. Note that with using an AprilTag calibration board it is not necessary to always have the whole board in the field of view.

EmiyaEstelle, 2019-10-11:
Thanks for your advice. @floriantschopp 
The resolution is 752×380. I will try it again to cover the whole image.
Besides, how do you see the coverage condition? is it that the detctions are only some part of the whole picture in the left picture of each camera?

floriantschopp, 2019-10-11:
Ok for this resolution you should be able to get a reprojection error of ~0.3-0.5 px. And yes, the left part shows your coverage. Make sure you have good illumination to minimize motion blur and check the quality of the calibration board (flatness and size).

floriantschopp, 2019-10-18:
I only have very limited experience with hand-printed calibration boards. For our calibrations, we typically use a high-quality calibration board with tag sizes >8cm on a very hard surface (e.g. plexiglass) to ensure flatness. Furthermore, we use extensive lighting form high-power LED strips along both sides of the calibration board to guarantee good and uniform illumination.

EmiyaEstelle, 2019-10-21:
@floriantschopp 
Thanks for your reply in a long time. Actually,  every time Kalibr calibrates my camera, the baseline of initializing is about 140mm. I try another tool just like image_pipelin from ROS and CamOdoCal (from VINS). They can estimate the basline about 120mm.

goldbattle, 2022-06-02:
Closing due to age. Reopen with a bag if this is still an issue, thanks!

closed 2022-06-02 by goldbattle