ethz-asl/kalibr #311: How to put into a known intrinsics to calibrate the extrinsics?

PowerfulMoney, 2019-10-18:
How to put into a known intrinsics to calibrate the extrinsics?

MohrMa2, 2019-10-23:
That is a good point. I also would like to decouple the computation of intrinsic and extrinsic, as we do in our setups. I found that even using the same data, the intrinsics come completely different, shifting of the principle points by 50 pixels (using 1820 resolution and a 95 degree lens). 

I don't think it is currently implemented as such, so it is up to you:
The two main things which have to be done:
1.) you have to convert the intrinsics to a format used by kalibr
2.) modifiy kalibr, that it reads the given intrinsics and use it lateron. 

I have not checked if 2) is possible. I think it should be, I cannot imagine, kalibr is optimizing globally 
AND the intrinsics at the same time.

PowerfulMoney, 2019-10-23:
> That is a good point. I also would like to decouple the computation of intrinsic and extrinsic, as we do in our setups. I found that even using the same data, the intrinsics come completely [...]

I found that the intrinsics and extrinsics keep changing while optimizing.
Have you ever edit the kalibr code for this aim?
Welcome to discuss.

kannidekan, 2020-01-11:
execuse me, do you know how could i get the IMU YAML file of the requirements in the camera-imu calibration? #341

Thanks very much!

goldbattle, 2022-04-27:
This is currently not possible. See https://github.com/ethz-asl/kalibr/issues/23