ethz-asl/kalibr #315: Error in using CDE pckage

adityapande-1995, 2019-10-23:
Extracted the kalibr-cde package and tried to run camera and imu calibrations files. I'm running on Ubuntu 18.04, with a python virtualenv. Can someone help me out with this error please ? What package is it searching for exactly ?

**Terminal output:** 
`./kalibr_calibrate_imu_camera 
importing libraries
ImportError: numpy.core.multiarray failed to import
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/apport_python_hook.py", line 64, in apport_excepthook
    from apport.fileutils import likely_packaged, get_recent_crashes
ImportError: No module named apport.fileutils

Original exception was:
ImportError: numpy.core.multiarray failed to import
Traceback (most recent call last):
  File "/var/kalibr-build/devel/bin/kalibr_calibrate_imu_camera", line 5, in <module>
    exec(fh.read())
  File "<string>", line 3, in <module>
  File "/var/kalibr-build/devel/lib/python2.7/dist-packages/sm/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 1, in <module>
  File "/var/kalibr-build/devel/lib/python2.7/dist-packages/numpy_eigen/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 3, in <module>
ImportError: numpy.core.multiarray failed to import
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/apport_python_hook.py", line 64, in apport_excepthook
    from apport.fileutils import likely_packaged, get_recent_crashes
ImportError: No module named apport.fileutils

Original exception was:
Traceback (most recent call last):
  File "/var/kalibr-build/devel/bin/kalibr_calibrate_imu_camera", line 5, in <module>
    exec(fh.read())
  File "<string>", line 3, in <module>
  File "/var/kalibr-build/devel/lib/python2.7/dist-packages/sm/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 1, in <module>
  File "/var/kalibr-build/devel/lib/python2.7/dist-packages/numpy_eigen/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 3, in <module>
ImportError: numpy.core.multiarray failed to import`

andersfagerli, 2019-10-27:
Had the exact same import error when using the CDE package, and got nowhere after several hours of troubleshooting. Build from source instead, works fine in ROS Melodic.

EmreOzkose, 2019-12-21:
Hi @adityapande-1995 , Could you find any solution? I have the same problem now.

kannidekan, 2020-01-11:
execuse me, do you know how could i get the IMU YAML file of the requirements in the camera-imu calibration? #341

Thanks very much!

kaplanitay, 2020-07-25:
Hi
Having the same issue.
Any updates regarding this?
The only way to run this is to build from source?

Hellmett, 2020-09-15:
Same problem!!! any update please let me konw TAT

mhc994, 2020-09-18:
sudo apt-get install python-apport

then start a new terminal and run kalibr.

buenos-dan, 2020-11-21:
same problem. any suggestion?

YujiElfahkrany, 2021-01-16:
solved by following this https://github.com/tensorflow/tensorflow/issues/559#issuecomment-168534197

sbillin, 2021-11-18:
I found this happens when the topic name is mis-spelled for the "--topics" command-line argument

Solin32, 2022-03-16:
> Same problem!!! any update please let me konw TAT

Did you add title  `timestamp,omega_x,omega_y,omega_z,alpha_x,alpha_y,alpha_z` to imu0.csv? I had this problem solved after adding that.

goldbattle, 2022-04-22:
Build from source or use the docker.

closed 2022-04-22 by goldbattle