ethz-asl/kalibr #317: Build Problem / Ubuntu 18.04 / Melodic

HartnettMatt, 2019-10-29:
I'm trying to use catkin_build when building from source, and everything seems to go as planned until it hits 68% at which this shows up in terminal:

[ 68%] Built target numpy_eigen
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

Additionally, since the package shows up in my workspace and I can run the executables, I've tried to run:

 kalibr_calibrate_cameras --models pinhole-equi pinhole-equi --topics /usb_cam/image_raw --bag static.bag --target aprilgrid_6x6.yaml

To run a basic calibration of my webcam, but then this error appears:

ImportError: No module named libsm_python

I've tried deleting my workspace and re-building, but no results. Any help or advice would be appreciated.

MohrMa2, 2019-11-22:
I think the best option is to create a docker image based on Ubuntu 14 and ros.
Kailbr is based on Ubuntu14. I only get problems with 

I have attached a Dockerfile that I used. Rename it to Dockerfile (without .txt)
[Dockerfile.txt](https://github.com/ethz-asl/kalibr/files/3880047/Dockerfile.txt)
You have to use nvidia-docker. nvidia-docker1 is enough.

I am starting it with the following comands:
sudo xhost +local:root # 
nvidia-docker run --privileged -ti --rm -e DISPLAY=$DISPLAY --env="QT_X11_NO_MITSHM=1" -v /tmp/.X11-unix:/tmp/.X11-unix:rw  -v /home/me/KALIBR/:/KALIBR/ -e /bin/bash --network=host nv-kalibr-indigo
(Here the image is named nv-kalibr-indigo).

oafolabi, 2019-11-25:
@HartnettMatt , I ran into the same issue. I reverted to using python2. I think the problem with my set up was that I had anaconda installed and so its python kept getting called as the default python. When I switched to python two on my system without anaconda, it was fine.

To switch you can change the line "#!/usr/bin/env python" to "#!/usr/bin/env python2"  at the beginning of kalibr_workspace/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_cameras.py . Hope that helps

HartnettMatt, 2019-11-27:
So the actual problem was that the kalibr dependencies were not being properly built - I still don't know why, but the solution was to individually build the dependency that was stopping the catkin_make, then rebuild the whole project, rinse and repeat.

closed 2019-11-27 by HartnettMatt