ethz-asl/kalibr #318: Help!!something wrong with kalibr_create_target_pdf

a412841321, 2019-10-30:
When i execute " kalibr_create_target_pdf --type apriltag --nx 6 --ny 6 --tsize 0.08 --tspace 0.3
", error act like bellow.Who can help me,thx!
raceback (most recent call last):
  File "/home/chirs/Calibration/kalibr_workspace/devel/bin/kalibr_create_target_pdf", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/chirs/Calibration/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_create_target_pdf", line 206, in <module>
    generateAprilBoard(canvas, parsed.n_cols, parsed.n_rows, parsed.tsize, parsed.tagspacing, parsed.tagfamiliy, parsed.skipIds)
  File "/home/chirs/Calibration/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_create_target_pdf", line 122, in generateAprilBoard
    c.text(pos[0]+tagSize*0.3, pos[1], "x")
  File "/home/chirs/.local/lib/python2.7/site-packages/pyx/canvas.py", line 324, in text
    return self.insert(self.texrunner.text(x, y, atext, *args, **kwargs))
  File "/home/chirs/.local/lib/python2.7/site-packages/pyx/text.py", line 1194, in text
    self.execute(expr, self.defaulttexmessagesdefaultrun + self.texmessagesdefaultrun + texmessages)
  File "/home/chirs/.local/lib/python2.7/site-packages/pyx/text.py", line 951, in execute
    self.defaulttexmessagesstart + self.texmessagesstart)
  File "/home/chirs/.local/lib/python2.7/site-packages/pyx/text.py", line 1005, in execute
    self.texinput.write(self.expr)
IOError: [Errno 32] Broken pipe
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/home/chirs/.local/lib/python2.7/site-packages/pyx/text.py", line 748, in _cleantmp
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/home/chirs/.local/lib/python2.7/site-packages/pyx/text.py", line 663, in run
    raise RuntimeError("TeX/LaTeX finished unexpectedly")
RuntimeError: TeX/LaTeX finished unexpectedly

    texrunner.texinput.write("\n\\end\n")
IOError: [Errno 32] Broken pipe
Error in sys.exitfunc:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/home/chirs/.local/lib/python2.7/site-packages/pyx/text.py", line 748, in _cleantmp
    texrunner.texinput.write("\n\\end\n")
IOError: [Errno 32] Broken pipe

kannidekan, 2020-01-11:
execuse me, do you know how could i get the IMU YAML file of the requirements in the camera-imu calibration? #341

Thanks very much!

seanXYZ, 2022-02-21:
try python3

goldbattle, 2022-04-27:
Please fix your latex install and also try python3 through this PR: https://github.com/ethz-asl/kalibr/pull/515

closed 2022-04-27 by goldbattle