ethz-asl/kalibr #322: Reference Frames used in the calibration

frme92, 2019-11-06:
Hi, I'm a student using Kalibr with a Pointgrey camera and a 3DR Pixhawk Imu. 
Could someone tell me which reference frames are used in the calibration?
From the information found, it should be the Imu reference frame oriented with z-axis downwards, x-axis backward and y-axis to the left. 
Instead the one of the camera has a z-axis directed along the axis of the optics while the x-axis towards the right and the y-axis downwards.
Finally the world reference frame has x and y axis in the target plane with the first to the right and the second to the top. And then the z-axis coming out of the plane. it's correct?
If yes, what sequence does the rotation matrix T_ci use? 3-2-1 or 1-2-3?

kannidekan, 2020-01-11:
execuse me, do you know how could i get the IMU YAML file of the requirements in the camera-imu calibration? #341

Thanks very much!

frme92, 2020-01-11:
Yeah, as shown in [https://github.com/ethz-asl/kalibr/wiki/yaml-formats](url), what you're looking for is a .yaml file like this:

#Accelerometers
accelerometer_noise_density: 1.86e-03   #Noise density (continuous-time)
accelerometer_random_walk:   4.33e-04   #Bias random walk

#Gyroscopes
gyroscope_noise_density:     1.87e-04   #Noise density (continuous-time)
gyroscope_random_walk:       2.66e-05   #Bias random walk

rostopic:                    /imu0      #the IMU ROS topic
update_rate:                 200.0      #Hz (for discretization of the values above)

If you're lucky you can find those values in your IMU data sheet otherwise you need to do an Allan Variance as shown in [https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model](url). 
In the latter case, if you have a Matlab license there's a very useful toolbox in this page: [https://github.com/rpng/kalibr_allan](url)